me-no-dev/ESPAsyncWebServer #233: Answer port 80 for any address?

treii28, 2017-09-23:
Sorry if this is the wrong place to post such a question, but I've been trying to find answers all over the net and have come up empty on all my posts and queries.
Apparently the examples that are out there for captive portal on ESP8266 no longer work on newer phones.  The method I've seen posted involves setting up the dnsServer code to respond with the softAP address for any inquiry, then to answer the various connectivity-test urls used by google, microsoft, apple, etc.  I've tried these with multiple response types (blank 200, simple html 200 with redirect in the header, 302 redirect, the content I was redirecting to) and had no luck.
As of my tests, I was never even seeing the android requests for /generate_204 on the web server at all so I tracked down a modified dnsServer and 'could' see the requests for the google host name as well as see the fact the dnsServer was responding with the softAP ip address. So apparently android will fall back to a cached dns record or something or otherwise not trust the response in some cases. (because it's a 10.10.10.* network perhaps?)
 I am going to try running wireshark this weekend to get a better look at what's going on, but at this point I'm suspecting the cached dns theory is probably accurate and the android OS is requesting the /generate_204 from an address other than the softAP.
I haven't delved into deconstructing protocols or packets in decades and my knowledge of C/C++ is limited, albeit growing. I'm wondering if it's possible in some way to do something similar to linux IPTable rules to have the AsyncWebServer answer those /generate_204 and other urls regardless of the address they are requested from on the ESP8266? Examples helpful or at least references where I can start reading to learn how please!

My goal is that for the application I have in mind, I am not going to want to give people detailed instructions on how to 'find' the content on the ESP8266. I'm trying to set up something to use as a remote wifi-geocache node where users can try to find or at least get close to the device, then use their cell phone to both know when they are in fact close, then sign into a simple web-page with a simple logging script to prove they were there and in fact found it.  Ultimately, some of the people that may potentially use it will never have made contact with me at all, so if I can get their device to trigger the pop-up to go to a landing page (similar to what some routers do on public wifi hotspots like at McDonald's) that either included the log-sign-in page or a link to it, that would be about the only way I can make such a thing feasible and practical.

me-no-dev, 2017-09-25:
Hmmmm... I have not looked recently at captive either. You can detect if the request is coming on the SoftAP interface as long as the DNS server tells it that that is where the site is located. If you do a more verbose DNS queries and use onNotFound in AsyncWebServer (with printing everything about the request) maybe we can figure something out... also maybe you can set the SoftAP IP pool to something less local? Just throwing ideas...

treii28, 2017-09-29:
re: "also maybe you can set the SoftAP IP pool to something less local? "

Pool? Can it be set up to respond to multiple addresses? Since I'm not connecting to the internet in any way with these devices, I considered trying to 'spoof' the addresses but didn't think the softAP could easily be set to answer to multiple addresses. Or are you talking about configuring more than one softAP?

me-no-dev, 2017-09-30:
what i mean by that is to set ap ip and dhcp pool to non-local ip addresses. not 192.168.x.x but something that would be a real external ip.

closed 2019-10-05 by stale[bot]