me-no-dev/ESPAsyncWebServer #234: Cant get library to work

timstans, 2017-09-24:
when sending a http request the esp32 crashes with the error below

Guru Meditation Error of type InstrFetchProhibited occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x00000000  PS      : 0x00060630  A0      : 0x800d63b2  A1      : 0x3ffe21a0  
A2      : 0x3ffc3584  A3      : 0x3ffe21f0  A4      : 0x00000003  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x800d62b2  A9      : 0x3ffe2180  
A10     : 0x3ffc3584  A11     : 0x00000001  A12     : 0x3ffdfabc  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000000  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x00000000:0x3ffe21a0 0x400d63af:0x3ffe21c0 0x400d421d:0x3ffe21f0 0x400d4256:0x3ffe2220 0x400d321b:0x3ffe2260 0x400d322e:0x3ffe2280 0x400d4523:0x3ffe22a0 0x400d3987:0x3ffe22e0 0x400d39f1:0x3ffe2320 0x400d3bd9:0x3ffe2360 0x400d2021:0x3ffe2380 0x400d21fd:0x3ffe23d0

anybody know how to fix?

me-no-dev, 2017-09-25:
So you basically gave us no info at all... where is your code? how are you setting things up? Why not decode that exception (https://github.com/me-no-dev/EspExceptionDecoder)? Please give more info or there is really nothing that anyone can do.

timstans, 2017-09-25:
Decoding 13 results
0x400d70af: __cxa_guard_release at /Volumes/build/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/guard.cc line 400
0x400d4e25: DefaultHeaders::Instance() at /Users/tim/Documents/Arduino/hardware/espressif/esp32/libraries/ESPAsyncWebServer/src/WebResponses.cpp line 494
:  (inlined by) AsyncWebServerResponse::AsyncWebServerResponse() at /Users/tim/Documents/Arduino/hardware/espressif/esp32/libraries/ESPAsyncWebServer/src/WebResponses.cpp line 98
0x400d4e5e: AsyncBasicResponse::AsyncBasicResponse(int, String const&, String const&) at /Users/tim/Documents/Arduino/hardware/espressif/esp32/libraries/ESPAsyncWebServer/src/WebResponses.cpp line 494
0x400d3e23: AsyncWebServerRequest::beginResponse(int, String const&, String const&) at /Users/tim/Documents/Arduino/hardware/espressif/esp32/libraries/ESPAsyncWebServer/src/WebRequest.cpp line 928
0x400d3e36: AsyncWebServerRequest::send(int, String const&, String const&) at /Users/tim/Documents/Arduino/hardware/espressif/esp32/libraries/ESPAsyncWebServer/src/WebRequest.cpp line 928
0x400d512b: AsyncCallbackWebHandler::handleRequest(AsyncWebServerRequest*) at /Users/tim/Documents/Arduino/hardware/espressif/esp32/libraries/ESPAsyncWebServer/src/WebServer.cpp line 79
0x400d458f: AsyncWebServerRequest::_parseLine() at /Users/tim/Documents/Arduino/hardware/espressif/esp32/libraries/ESPAsyncWebServer/src/WebRequest.cpp line 561
0x400d45f9: AsyncWebServerRequest::_onData(void*, unsigned int) at /Users/tim/Documents/Arduino/hardware/espressif/esp32/libraries/ESPAsyncWebServer/src/WebRequest.cpp line 121
0x400d47e1: std::_Function_handler ::_M_invoke(std::_Any_data const&, void*&&, AsyncClient*&&, std::_Any_data const&, unsigned int&&) at /Users/tim/Documents/Arduino/hardware/espressif/esp32/libraries/ESPAsyncWebServer/src/WebRequest.cpp line 75
:  (inlined by) _M_invoke at /Users/tim/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/functional line 1871
0x400d2c4d: std::function ::operator()(void*, AsyncClient*, void*, unsigned int) const at /Users/tim/Documents/Arduino/hardware/espressif/esp32/libraries/AsyncTCP/src/AsyncTCP.cpp line 660
:  (inlined by) AsyncClient::_recv(tcp_pcb*, pbuf*, signed char) at /Users/tim/Documents/Arduino/hardware/espressif/esp32/libraries/AsyncTCP/src/AsyncTCP.cpp line 497
0x400d2e29: AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, signed char) at /Users/tim/Documents/Arduino/hardware/espressif/esp32/libraries/AsyncTCP/src/AsyncTCP.cpp line 660
:  (inlined by) _handle_async_event at /Users/tim/Documents/Arduino/hardware/espressif/esp32/libraries/AsyncTCP/src/AsyncTCP.cpp line 80
:  (inlined by) _async_service_task at /Users/tim/Documents/Arduino/hardware/espressif/esp32/libraries/AsyncTCP/src/AsyncTCP.cpp line 96

timstans, 2017-09-25:
received when using the following code:

#include <ESPmDNS.h>
#include <WiFi.h>
#include "FS.h"
#include "SPIFFS.h"
#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>

AsyncWebServer server(80);

void onRequest(AsyncWebServerRequest *request){
  //Handle Unknown Request
  request->send(404);
}

void setup() {
  Serial.begin(115200);
  if(!SPIFFS.begin()){
        Serial.println("SPIFFS Mount Failed");
        SPIFFS.begin(true);
        ESP.restart();
    }
  WiFi.begin("SSID", "PASS");
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("WiFi Connected.");
  WiFi.setHostname("EZTile");
  Serial.print("IP Address: ");
  Serial.println(WiFi.localIP());
  if (!MDNS.begin("EZTile")) {
        Serial.println("Error setting up MDNS responder!");
        while(1) {
            delay(1000);
        }
   }

  MDNS.addService("_http", "_tcp", 80);

  server.on("/heap", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send(200, "text/plain", String(ESP.getFreeHeap()));
  });
  server.begin();
}

void loop() {
  

}

timstans, 2017-09-25:
Pretty formatted: https://hastebin.com/mavicijove.cpp

me-no-dev, 2017-09-26:
pretty formated above :) sketch looks good. will give it a shot

timstans, 2017-10-04:
Any Luck? @me-no-dev

s00500, 2017-10-05:
Seems like I stumbled into the same problem...

s00500, 2017-10-16:
Ok, my holidays are over: The Problem seems to be the new default header feature by tetious, merged on 12th September, in WebResponses.cpp Line 97:

```
{
  for(auto header: DefaultHeaders::Instance()) {
    _headers.add(new AsyncWebHeader(header->name(), header->value()));
}
}
```

When I comment this out it works...

s00500, 2017-10-27:
Any news on this issue?

s00500, 2017-11-08:
@me-no-dev  Please change at least the titel...

me-no-dev, 2017-11-08:
sorry guys :( I have been busy with work (and I am in the process of moving also). what should I do to crash this sketch? because I can't :)

s00500, 2017-11-08:
Np It does on the ESP32... I can test it again on friday and report here, maybe change the title of the issue to something more explicit though 😅

me-no-dev, 2017-11-08:
to what though 🙄 🤣

s00500, 2017-11-08:
Something like "ESP32 Possible crash with default header feature" ? 🙃

me-no-dev, 2017-11-08:
done 😝

me-no-dev, 2017-11-08:
so I am on a Mac. I tried both Chrome and Safari to open known and unknown pages on the sketch above. No hiccup so far

s00500, 2017-11-08:
Can't test it right now but I will this week again, I encountered the issue with the exact same exception on a mac using chrome but I don't think it has anything to do with the browser

s00500, 2017-11-09:
Ok very strange, I also cannot replicate it right now... tested with and without spiffs @timstans if you agree this can be closed =D

s00500, 2017-11-13:
@me-no-dev I think you can close it

closed 2017-11-13 by me-no-dev