me-no-dev/ESPAsyncWebServer #235: Recursive Response

Kaisha, 2017-09-25:
So I decided to go ahead and write a recursive response class for ESPAsync.  I have a few little things I need to clean up, and I haven't properly tested it yet (just a few basic tests); but it seems to be working.  That said, I'm wondering what would be the best way to incorporate it into ESPAsync.

The general idea is that you pass a string with templates, and a callback which returns a string associated with each template.  The main difference is that this string that is returned is then also parsed for templates, recursively.  So you can construct large dynamic web pages with low memory usage.

I also added a StringSource and StringWrapper class.  The internals of ESPAsync uses a lot of String manipulation/String objects, which are dynamically allocated on the heap.  Most of the time this is unnecessary, and where (I feel) many of the the out of memory problems are coming from.  StringSource and StringWrapper allow RecursiveResponse to use very little dynamic memory, only allocating when absolutely necessary.

I also added a few progmem helper routines, so working with progmem should be faster than the standard ESP8266 ones.

So... at this point I'm kinda of curious as to what people would like me to do with this (in particular me-no-dev).  Should I fork it?  Should I upload the files somewhere?  Is there an interest in incorporating it or is there no interest at all?  Basically, where should I go from here?

me-no-dev, 2017-09-25:
Your best bet is to fork the repository, incorporate the feature and make a pull request so we can have a look, give it a go and comment on the changes :)

Kaisha, 2017-09-25:
Ok, I've never used git before, but I think I've managed to fork, upload the changes, and send a pull.  If it didn't work let me know.

There were a few things that I was entirely certain how people would like them to be, or how they were supposed to work, and a few issues as well.  I've documented most of it in the source, but in summary:

- The _ack callback has 'n' and 'time' but I wasn't sure what they were for.
- I've read in places that i/o stream isn't well supported in Arduino, so I didn't integrate it, I'm not sure if people want stream integration, there's also StringStream.h and Stream.h from Arduino, TBH I'm not sure how they differ from C++ streams and if people would like to see StringSource/StringWrapper integration.
- In ESPAsyncWebServer you include Arduino.h prior to functional.  Sometimes this can cause compilation errors due to macro conflicts (min/max).  I found it best to include all standard libraries (ie. cstdint, functional, etc...) before including Arduino libraries.
- As it stands, to keep things simple, there is only one callback.  If people are interested, it would be possible to extend this to allow for recursive callbacks (ie. requiring a callback to be returned along with the StringWrapper).  It sounds complicated, but it really isn't once you see it in action, and would make things much easier.

Kaisha, 2017-09-26:
Added recursive callbacks.  So now each level can have a unique call back too.  Coupled with std::bind() this really makes passing data around easy.
- I'm wondering if I should remove StringWrapper and roll it into RecursiveNode, would require a small change to member functions setContentType() and addHeader()

me21, 2017-10-25:
@Kaisha, just a note: with current template engine you can call template processor function recursively yourself if you want to parse the returned string for templates recursively.

dfelsch, 2018-10-14:
Can you give a small example how to call the template processor function directly?

closed 2019-10-05 by stale[bot]

me21, 2020-02-02:
@dfelsch just call the function from inside itself, recursively