me-no-dev/ESPAsyncWebServer #237: esp32 compile fails with platformio

andig, 2017-09-30:
platformio run --environment esp8266
	Compiling .pioenvs/esp8266/lib/ESPAsyncWebServer_ID306/WebAuthentication.o
	.piolibdeps/ESPAsyncWebServer_ID306/src/AsyncWebSocket.cpp:40:18: fatal error: Hash.h: No such
	 file or directory
	#include <Hash.h>
	^
	compilation terminated.
	*** [.pioenvs/esp8266/lib/ESPAsyncWebServer_ID306/AsyncWebSocket.o] Error 1

	========================== [ERROR] Took 28.15 seconds ==========================
	Environment esp8266     [ERROR]
	========================== [ERROR] Took 28.15 seconds ==========================

platformio.ini:

	[env:esp8266]
	platform=espressif8266
	framework=arduino
	board=esp12e

It seems `Hash.h` doesn't exist on the platformio esp8266 platform?

me-no-dev, 2017-10-01:
@andig maybe ping @ivankravets though I am pretty sure that all is fine on both ESP8266 and ESP32 :) we would have known by now if something was missing.

andig, 2017-10-01:
> I am pretty sure that all is fine on both ESP8266 and ESP32 :) we would have known by now if something was missing.

That's what I've thought. But even reinstalling atom/pio ide leads to the same issue. Would you happen to know where the actual platform sources are installed so I could check for presence of hash.h (as its in the esp8266 git repo)?

me-no-dev, 2017-10-01:
no idea really :) i kinda need to use only the git version of both repos. You could serach for Arduino.h your computer though

andig, 2017-10-01:
I've had another look. It is working if I use this:

	lib_deps =
	  Hash
	  ESPAsyncWebServer

Without explicitly specifying hash (as build-in library) it fails. So that's either a behaviour change in pio 3.5 (and an enhancement here) or a bug in pio?

me21, 2017-11-23:
@andig Perhaps it's Platformio Library Dependency Finder (LDF) working mode? I.e. it works in standard mode, looking for lib dependencies only in headers, whereas it should work in deep mode, looking for lib dependencies in source files too.

andig, 2017-11-23:
Opened various LDF tickets at PIO, can close here.

closed 2017-11-23 by andig