me-no-dev/ESPAsyncWebServer #238: Improvement: Add user data pointer to AsyncWebSocketClient

herm, 2017-09-30:
Often it is required to cache a full websocket message before processing it. This buffer needs to be associated to an individual client to avoid race conditions. Currently I need to use std::map to map the client id to a buffer. It would be faster and easier if there was a void * in  AsyncWebSocketClient which could be used for any purpose the user of this library wants.

Many other libraries do the same thing, either providing a pointer to the callback or allowing to store data on the object.

Another option would be a new variant of AsyncWebSocketClient which does the buffering internally and only calls the callback once the message is completely received.

me-no-dev, 2017-10-01:
Done ;) use _tempObject for that *the same one exists also for WebRequest)

closed 2017-10-01 by me-no-dev

herm, 2017-10-01:
Thank you for the quick response. Just one remark: If this is public API the name should probably start without an underscore.

me-no-dev, 2017-10-02:
maybe but it already does in the request class so better have it the same :)