me-no-dev/ESPAsyncWebServer #244: Web server stop responding with use of delay() in main loop

joehui, 2017-10-02:
In the following example code, I use AsyncWebServer to serve an HTML page (URL is /ui), and have a function to respond to an Ajax call (URL is /b).   These HTTP calls start out fine immediately after bootup of ESP8266.  But, after a few seconds, it will start to respond very very slowly or stop responding entirely.  If I remove the delay() call in the main loop, then everything works fine.  I suspect I must have done something wrong.  I already spent many hours on this, but can't figure it out.

I am building with ESP8266 library version 2.3.0.

Here's the example code:

```
#include <ESP8266WiFi.h>
#include <ESPAsyncTCP.h>
#include <ESPAsyncWebServer.h>

const char* ssid = "Joey2.4G";
const char* password = "selinahui";

AsyncWebServer httpServer(80);
volatile short globalValue = 5;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(250000);
  
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  if (WiFi.waitForConnectResult() != WL_CONNECTED)
    Serial.println("WiFi failed, retrying.");

  // REST function to be called by AJAX
  httpServer.on("/b", HTTP_GET, [](AsyncWebServerRequest *request){
    AsyncWebParameter* p = request->getParam("index");
    globalValue = strtol(p->value().c_str(), NULL, 10);        
    request->send(200, "text/plain", "");
  });
  

  // HTML page
  httpServer.on("/ui", HTTP_GET, [](AsyncWebServerRequest *request){
    char htmlText[2000];
    
    sprintf(htmlText, "<HTML><body>\n \
      <meta name=\"viewport\" content=\"width=200\">\n \
      <input id=\"bindex\" type =\"range\" min =\"1\" max=\"13\" step =\"0.4\" value =\"%d\"/>\n \
      <script>\n \
        var b = document.getElementById(\"bindex\");\n \
        b.addEventListener(\"input\", function() {\n \
          var xhReq = new XMLHttpRequest();\n \
          xhReq.open(\"GET\", \"b?index=\"+ b.value, false);\n \
          xhReq.send(null);\n \
        }, false);\n \
      </script>\n \
      </body></HTML>", globalValue);
      
    request->send(200, "text/html", htmlText);
  });

  // start web server
  httpServer.begin();  
}

void loop() {
  // put your main code here, to run repeatedly:
  Serial.println("main loop");
    
  delay(100);
}
```

me-no-dev, 2017-10-03:
code looks fine to me. no clue really. i need to look into delay a bit to figure out, but i am away from my computer

joehui, 2017-10-03:
Thanks for your speedy reply.  I have a bit little more information.

The issue seems to be specific to the wifi router I am using.  I tested this code in the office, I couldn't reproduce the problem any more.

I will do more investigation when I get home.  I will see if I could fix this by changing some router settings.  But, it is a bit strange that the use of delay() would have an effect on this.

joehui, 2017-10-03:
It is definitely something to do with the settings on my Router, which had a beacon interval of 500ms, and DTIM value of 150.  I think this means that the wifi client would only get a beacon once every 75 seconds.

The delay() call in ESP8266 probably have caused it to think it can rest and thus trigger the wifi component to go into sleep mode.  The wifi component is awaken only on the next receipt of the beacon. That's why once delay() is called, the web server stops responding for more than a minute.  In fact, the whole ESP is unreachable during this time, not just the web server.

By reducing the beacon interval to 100ms and DTIM value to 5, everything now works smoothly.

So, I believe this issue is nothing to do with AsyncWebServer.

closed 2017-10-03 by joehui