me-no-dev/ESPAsyncWebServer #245: Error on load CSS and JS Files from browser

rick-bp, 2017-10-02:
Hello ME-NO-DEV, you're the guy!
I'm using ESPAsyncWebServer in a complex project.
I'll try to explain superficially.
One ESP12E is used as Access Point to others ESP12E Stations.
Each device (AP and stations) have inside the Filesystem many files like .html, .css, .js, .json, .png, and others file type.
When a static webpage is called from browser, the page call others files eg: images, CSS, JS (JQuery, bootstrap, ...).
The problem is: sometimes, the browser canot get download the others files to render the webpage.
I am working around this eg embeding the files on the main file, but, I and you know that isn't a best pratice.
The maintenance are hard and each adjust I have to do in all files that use the external file embedding.

Did you see anything like my problem?
Any tip to me?

PS. I don't use delay() in the project.
If I press <F5> in the browser some times, the page is loaded.

Sorry for my English!
[image]
[image]
[image]

me-no-dev, 2017-10-03:
some browsers do not handle well outstanding requests. since we have a limit of 5 concurent connections, when they increase above that it’s possible that the browser will close some other outstanding ones and do not load your files. embedding in this case is the bst choice :)

rick-bp, 2017-10-03:
:/

I Have some questions...

the 5 concurrent connections are defined by?

There is a queue of requests?

Thinking here...
Template Processor can be another choice?

I did embed all the images on the main files. [image]

but, only the bootstrap.min.css + bootstrap.min.js + jquery-3.2.1.min.js = 256 KB (without gziped)

I will continue to think on a solution... If you have some idea please, tell me!
I think there are something wrong, because, the main files call between  3 ~ 5 files after embeded images.

Thanks a lot!

me-no-dev, 2017-10-04:
5 connections are limit in the esp network stack itself. the async implementation does not limit anything. you can use the browsers dev console to check if request is canceled by the browser. templates are kinda solution but there are so many options for templates... kinds not doable for all options... i have also been thinking quite a bit for that. gzipping and embedding so far is the best route. also using external resources which surely do not work over AP interface sin its local and does not provide routing to outside STA

rick-bp, 2017-10-04:
Thanks for reply!
I did embeded all the images as src="base64" and the scripts jquery.js, proper.js, bootstrap.js, wnumb.js, nouislider.js and canvas gauge.js in a single gzipped file.
For the moment, I think this like have been resolved, but I'd like other solution.
I tried to load files with ajax, but does not worked too.
The best solution for this moment realy is embebed files.
I have no connection with external internet in this devices. Then I canot load files from others servers.
I'm curious, why me-no-dev? kkkk
Have a nice day!

me-no-dev, 2017-10-04:
You can use ajax and load the files one after another. Just create an ajax loader class that queues the requests and executes them one by one. That would give you good performance as well. I think I have this somewhere in a file but do not remember where and do not have my computer with me

rick-bp, 2017-10-06:
I Think I found a issue on lib. I am investigating, and return if true.

wahabkhan1892, 2017-10-11:
I'm having the same problem. Have you found any solution other than embedding the files?

wahabkhan1892, 2017-10-11:
I tried loading a single large file, and was getting the same problem. When I looked at Wireshark capture, the client was sending a [FIN, ACK] packet to the server prematurely, thus closing the port.

rick-bp, 2017-10-11:
wahabkhan1892,
I am still working on it!
In my case, embedding files are working at the moment, but, I am looking for an other solution.

closed 2019-10-05 by stale[bot]