me-no-dev/ESPAsyncWebServer #248: Response outside query function

mcmega, 2017-10-04:
Hello!
I send a request from the browser (/settings)
This request is processed and sent to the external MCU by UART. In the main loop, I get a response on the UART, I process it. Reply from the MCU I need to transfer back to the browser.
How to respond to such a request, in which it is not necessary to respond immediately, but first to obtain data from the MCU via UART?

Sorry for English :)

mcmega, 2017-10-06:
Can not create a response outside the function.

me-no-dev, 2017-10-08:
use chunked response... is this ESP32 or ESP8266? Would not be possible on ESP8266. And you are not responding to the request in your handler above which will lead to the server breaking.

mcmega, 2017-10-08:
I use ESP8266 (8285).
I want to respond with data that comes from UART, so there is no answer to the query. But I do not know how to do it.

mcmega, 2017-11-26:
I did everything on WebSocket, but the question is not settled, it would not be bad to have more examples!

closed 2017-11-26 by mcmega