me-no-dev/ESPAsyncWebServer #250: Error: 'class AsyncWebServer' has no member named 'client'

mcmega, 2017-10-05:
Hello!
I want to run SSDP, but when compiling an error.

```
server.on("/description.xml", HTTP_GET, [](AsyncWebServerRequest *request) {
        SSDP.schema(server.client());
});
```

me-no-dev, 2017-10-05:
You seem to really be stuck on the regular ESP8266WebServer and I already told you that this is something totally different. if you want to run SSDP over the AsyncWebServer see how I have it implemented here: https://github.com/me-no-dev/esp8266fcgui/blob/master/esp8266fcgui.ino#L96-L115

closed 2017-10-05 by me-no-dev

mcmega, 2017-10-05:
Yes, it seems stuck)
Thanks, I will understand. Examples would be more ...

JoanMCD, 2018-07-18:
I have mod the `void uSSDP::schema(WiFiClient client)` to `String uSSDP::schema(void)` and use it with AsyncWebserver as follow:

```
  webserver.on("/description.xml", HTTP_GET, [](AsyncWebServerRequest * request)
  {
    request->send(200, "text/xml", SSDP.schema());
  });
```

uSSDP.cpp:
```
String uSSDP::schema(void) {
	String s = "";
    s +="<?xml version=\"1.0\"?>\n";
    s +="<root xmlns=\"urn:schemas-upnp-org:device-1-0\">\n";
    s +="\t<specVersion>\n";
    s +="\t\t<major>1</major>\n";
    s +="\t\t<minor>0</minor>\n";
    s +="\t</specVersion>\n";

    s +="\t<device>\n";
    s +="\t\t<deviceType>urn:schemas-upnp-org:device:Basic:1</deviceType>\n";

    if (strlen(_device->presentationURL())) {
        s +="\t<presentationURL>";
        s +=_device->presentationURL();
        s +="</presentationURL>\r\n";
    }

    s +="\t\t<friendlyName>";
    s +=_device->friendlyName();
    s +="</friendlyName>\r\n";

    s +="\t\t<modelName>";
    s +=_device->modelName();
    s +="</modelName>\r\n";

    version_t *modelNumber = _device->modelNumber();

    if (modelNumber->major > 0 || modelNumber->minor > 0) {
        s +="\t\t<modelNumber>";
        s +=modelNumber->major;
        s +=".";
        s +=modelNumber->minor;
        s +="</modelNumber>\r\n";
    }

    if (strlen(_device->serialNumber())) {
        s +="\t\t<serialNumber>";
        s +=_device->serialNumber();
        s +="</serialNumber>\r\n";
    }

    s +="\t\t<manufacturer>";
    s +=_device->manufacturer();
    s +="</manufacturer>\r\n";

    if (strlen(_device->manufacturerURL())) {
        s +="\t\t<manufacturerURL>";
        s +=_device->manufacturerURL();
        s +="</manufacturerURL>\r\n";
    }

    s +="\t\t<UDN>uuid:";
    s +=_device->uuid();
    s +="</UDN>\r\n";

    s +="\t</device>\n";
    s +="</root>\n";
    //client.stop();
	return s;
}
```

pit001, 2018-10-11:
there is a small memory hole in uDevice.cpp

FreeHeapKb: 58.23
FreeHeapKb: 58.23
FreeHeapKb: 58.23
FreeHeapKb: 58.23
FreeHeapKb: 58.23
FreeHeapKb: 58.23
FreeHeapKb: 58.23
FreeHeapKb: 58.23
FreeHeapKb: 58.11
FreeHeapKb: 58.12
FreeHeapKb: 58.12
FreeHeapKb: 58.12
FreeHeapKb: 58.12
FreeHeapKb: 58.12
FreeHeapKb: 58.12
FreeHeapKb: 58.12
FreeHeapKb: 58.12
FreeHeapKb: 58.06
FreeHeapKb: 58.06
FreeHeapKb: 58.06
FreeHeapKb: 58.06
FreeHeapKb: 58.06
FreeHeapKb: 58.06
FreeHeapKb: 58.06
FreeHeapKb: 57.51
FreeHeapKb: 57.41
FreeHeapKb: 57.41
FreeHeapKb: 57.41
FreeHeapKb: 56.86
FreeHeapKb: 56.75
FreeHeapKb: 56.75
FreeHeapKb: 56.48
FreeHeapKb: 56.48
FreeHeapKb: 56.48
FreeHeapKb: 56.48
FreeHeapKb: 56.48
FreeHeapKb: 56.48
FreeHeapKb: 56.48
FreeHeapKb: 56.48
FreeHeapKb: 56.48
FreeHeapKb: 56.48
FreeHeapKb: 56.48
FreeHeapKb: 56.48
FreeHeapKb: 56.48
FreeHeapKb: 56.48
FreeHeapKb: 56.48
```

Fix:

char *uDevice::uuid() {
    //char *_uuid = (char *) malloc(37); //old
	char _uuid[37]; //new
    sprintf(_uuid, "%s-%02X%02X%02X%02X%02X%02X", _base, _mac[0], _mac[1], _mac[2], _mac[3], _mac[4], _mac[5]);
    return _uuid;
}
```

Pablo2048, 2018-10-12:
@pit001 two things:
1. this is wrong repository - it belongs to here https://github.com/me-no-dev/uSSDP
2. its not a bug - your _uuid[37] is local variable on the stack and his scope is only within ::uid() so your "fix" is wrong

Proper solution is to use free() in higher level after calling ::uid() IMHO.

Lightwell-bg, 2019-12-31:
Small library for SSDP https://github.com/Lightwell-bg/ssdpAWS