me-no-dev/ESPAsyncWebServer #254: Templating problem

thedroid, 2017-10-10:
There seems to be an issue with the template processor as it does not always replace correctly.  It is 100% reproducible on my system with the test program attached.  Please see the screenshot for the problem.

using 
git ESPAsyncWebServer
git Eps8266 Arduino
NodeMCU .9

[test.zip](https://github.com/me-no-dev/ESPAsyncWebServer/files/1370257/test.zip)
[image]

me21, 2017-10-24:
I'll look at it. Meanwhile, it seems there's an alternative template engine in PR #236.

me21, 2017-10-25:
Try PR #267, it should fix the bug.

thedroid, 2017-10-25:
Will test and report back.

thedroid, 2017-10-25:
Will test and report back

closed 2017-10-25 by thedroid

aharshac, 2017-12-16:
@me21 @thedroid The problem still exists.

me21, 2017-12-16:
@aharshac please give an example. Or did you mean the problem still exists with thedroid's original test.zip?

PR #267 (2017-11-06): Fixed templating engine bug.
This fixes #254.