me-no-dev/ESPAsyncWebServer #256: External CSS and JS files not loading

wahabkhan1892, 2017-10-11:
I have an index.html file with links to several JS and CSS files, a couple of which are 54KB and 300KB. I am using ESP8266 in AP mode. When I connect to it and type in 192.168.4.1, I can see through Wireshark capture, that index.html is being loaded properly and after that, the client requests 5 new ports. The server acknowledges these requests. The client then sends GET requests for the JS and CSS files on these ports, which are also acknowledged by the server. But after that the server doesn't send anything, when its supposed to start sending data packets.
[image]
[image]

jeremypoulter, 2017-10-11:
This sounds very much like issue #194 I raised a while ago. How much memory do you have free? FYI, I ended up working around this issue by a combination of embedding all the resources in PROGMEM (and avoiding SPIFFS as that allocates a large chunk of memory), combining all JS/CSS files into a single file and only using a small buffer when reading from flash and sending on to AsyncWebServer, see https://github.com/OpenEVSE/ESP8266_WiFi_v2.x

wahabkhan1892, 2017-10-11:
Is there a way to reduce memory usage while keeping SPIFFS and all the files?

rick-bp, 2017-10-11:
I have the same problem.
#245

bah-sa, 2017-12-04:
I have the same problem.
Only one linked file in page (CSS) and it do not loading 8(

me-no-dev, 2017-12-04:
So all of you have this issue only in AP mode?

wahabkhan1892, 2017-12-04:
Apparently yes, only in AP mode. I managed to get around this by packing all external files into one index file. In that way, the client doesn't have to open extra sockets, and the device doesn't run out of memory.

bah-sa, 2017-12-05:
I have problem in STA mode.
But I not sure it is AsyncServer issue.
Linked stylesheet in page dont applied on any mobile browser (chrome, samsung, opera mini), but on PC MSIE successfully applied.
Logs show web server send file, but mobile client browser dont apply it.
Postman succesfully load it also, content-type in headers is correct 'text/css'.
8(

bah-sa, 2017-12-05:
Problem fixed. 8)
CSS-file must be in UTF-8 encoding with BOM (!!!), otherwise stylesheet not applied.

thomasgresch, 2018-02-25:
Had the same problem - combing the files worked for me to, but I would be interested in a more elegant solution.

vindolin, 2018-02-26:
@bah-sa strange, that doesn't seem to make sense.
Can you copy the request HAR with Chrome or Firefox and post it here?

closed 2019-10-05 by stale[bot]

briangraphted, 2020-02-26:
I had this problem too. My esp was loading index.htm, style.css, and 3 separate .js files. I spent time minifying all the files but that didn’t solve the issue. Then I combined two of the .js files so I had 4 downloads total (index + css + js+ js). Then it worked!  I believe the key here is in the (great) analysis by wahabhkah1892 mentioned above “the client requests 5 new ports”. The esp either can’t handle that many (hard coded connection pool?) or runs out of memory or both. My observed rule of thumb: limit your files to 4 (and do all the good hygiene of minimizing them, and if needed compress and embed them [in that order]). Good luck!