me-no-dev/ESPAsyncWebServer #257: ESP32 example wanted

emard, 2017-10-16:
I'd like if you can add some ESP32 example that can get
errorless compile from arduino

JimPos, 2017-11-20:
The existing example can be made to work on the ESP32 with the following changes.
1. change the include files for ESP32
#ifdef ESP32
#include <WiFi.h>
#include <ESPmDNS.h>
#include <ArduinoOTA.h>
#include <FS.h>
//#include <Hash.h>
#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <SPIFFS.h>
#include <SPIFFSEditor.h>
#endif //ESP32

2. at approximately line 138 of the example, change the line creating the SPIFFS editor
#ifdef ESP32
  server.addHandler(new SPIFFSEditor(SPIFFS,http_username,http_password));
#endif
#ifdef ESP8266
  server.addHandler(new SPIFFSEditor(http_username,http_password));
#endif

Don't forget to enter your own SSID and password, and load data files to SPIFFS

emard, 2017-12-10:
HI!

I followed your suggestions, got it to compile and work
I've written simple line to "index.htm" file using SPIFFS_Test
example with this modification

    #if ESP8266
    if(!SPIFFS.begin()){
    #endif
    #if ESP32
    if(!SPIFFS.begin(true)){
    #endif

and my line appears in home page of asyncfswebbrowser

Now I need more help on how to make spiffs from PC.
I tried that to create and uploaded SPIFFS with this

~/src/downloaded/ESP32_spiffs_example/components/mkspiffs/src/mkspiffs \
 -c $1 -b 8192 -p 256  \
 -s 1048576 $2

~/Arduino/hardware/espressif/esp32/tools/esptool.py \
--chip esp32 --port $1 --baud 921600 \
write_flash --flash_size detect 0x180000 $2

But it doesn't work
(content of FS doesn't change, as if I haven't done anything).

I've changed this according to your suggestions to make asyncfsweb work on
ESP32
#if ESP8266
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <ESPAsyncTCP.h>
#include <Hash.h>
#include <FS.h>
#endif
#if ESP32
#include <WiFi.h>
#include <ESPmDNS.h>
#include <AsyncTCP.h>
#include <SPIFFS.h>
#endif

  #if ESP8266
  SPIFFS.begin();
  #endif
  #if ESP32
  SPIFFS.begin(true);
  #endif

  #ifdef ESP8266
  server.addHandler(new SPIFFSEditor(http_username,http_password));
  #endif
  #ifdef ESP32
  server.addHandler(new SPIFFSEditor(SPIFFS,http_username,http_password));
  #endif

Best regards, Davor

emard, 2017-12-10:
HI Again!

Wait, now it works! I upgraded to the latest [espressif/arduino-esp32](https://github.com/espressif/arduino-esp32) tools and
downloaded [me-no-dev/arduino-esp32fs-plugin](https://github.com/me-no-dev/arduino-esp32fs-plugin)

Thanks!

me-no-dev, 2017-12-11:
BTW ```-b 8192``` was wrong :) 4096 is the correct number

closed 2017-12-11 by me-no-dev

emard, 2017-12-11:
Thanks!

If there's no problem with it, can you add ESP32/ESP8266 #defines
so that it just compiles out of the box