me-no-dev/ESPAsyncWebServer #259: WebSocket works on PC's Chrome but wont on Android's Chrome.

aguaviva, 2017-10-18:
First off, thanks for this amazing project :) Thanks to it I was able to build a sous vide controller.

https://github.com/aguaviva/SousVide

It turns out that it works fine on PC 
[image]

but on my mobile and my tablet I get this:
[image]

The main.cpp file in my github repo shows how am I using the websockets.

In the setup() I init the websockets like this:

    ws.onEvent([](AsyncWebSocket * server, AsyncWebSocketClient * client, AwsEventType type, void * arg, uint8_t *data, size_t len)
    {
      if(type == WS_EVT_CONNECT)
      {
          Serial.printf("ws[%s][%u] connected\n", server->url(), client->id());
          client->ping();
          websocketsOnConnect(client);
      }
      else if(type == WS_EVT_DISCONNECT)
      {
          Serial.printf("ws[%s][%u] disconnected\n", server->url(), client->id());
      }
    });
    server.addHandler(&ws);

and in the loop() I do something like this:

        if (DataStreamer.tick())
        {
            STATE st = { (float)Temperature, (float)Power };
            histData.Queue(st);

            ws.binaryAll((char*)&st, sizeof(STATE));
        }

Any ideas what could be going on? I'll be happy to test any experiments you may want me to test.

omersiar, 2017-10-19:
I also encountered that issue. In my case Android Chrome did not get lengthy messages (i tried only plain text) so I reduced the plain text length, and it worked. I do not know if there is limit. Chrome dev team should be contacted I guess.

aguaviva, 2017-10-19:
Other ESP websocket libs are working fine on android. It might be an issue of this lib.

aguaviva, 2017-11-05:
@me-no-dev I'm very gently bumping this in case you missed that WebSockets are not working on Android. Also please add my project to the list of projects featuring your great lib.

me-no-dev, 2017-11-06:
huh... can you please collect some more logs to establish a pattern? I see in the images above that 1 more byte is received on the tablet than on the PC.

nicbkw, 2017-11-08:
Hi @me-no-dev - also seeing this issue. I copied some of your example code and made it work on the Heltec-LoRa module. When configured as WiFi.mode(WIFI_AP_STA) I cannot access the website from either android browsers or IOS browsers. When configured as WiFi.softAP, access is OK from all browsers. Here is the relevant part of the code:

```
  WiFi.mode(WIFI_AP_STA);
  //WiFi.softAP(hostName);
  WiFi.begin(ssid, password);
  if (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.printf("STA: Failed!\n");
    u8x8.drawString(0, 1, "STA: Failed!");
    WiFi.disconnect(false);
    delay(1000);
    WiFi.begin(ssid, password);
  }

```

me-no-dev, 2017-11-08:
OooOoooOO what? So AP is somehow breaking it? What is only in AP mode?

nicbkw, 2017-11-08:
;-) not tried it - will do so now...

nicbkw, 2017-11-08:
Same issue with WiFi.mode(WIFI_STA). Android browser actually shows ERR_ADDRESS_UNREACHABLE. Works fine with Chrome on Mac. My Mac and PC are connected over ethernet. iPad and Android phone via WiFi from same network. Android phone can reach the router and display its webpage no problem.

me-no-dev, 2017-11-08:
so to recap, it's working only in WIFI_AP mode? That so does not make sense...

nicbkw, 2017-11-08:
This works fine, no other changes - using the ESP32 WiFi AP I can connect with both Android phone and iPad to the webserver at 192.168.4.1

```
  //WiFi.mode(WIFI_STA);
  WiFi.softAP(ssid, password);
  //WiFi.begin(ssid, password);
  //if (WiFi.waitForConnectResult() != WL_CONNECTED) {
    //Serial.printf("STA: Failed!\n");
    //u8x8.drawString(0, 1, "STA: Failed!");
    //WiFi.disconnect(false);
    //delay(1000);
    //WiFi.begin(ssid, password);
  //}

  u8x8.setCursor(0, 2);
  u8x8.print(WiFi.softAPIP());
```

nicbkw, 2017-11-08:
FYI - I am compiling the sketch using the arduino-esp32 from espressif and made a fresh install yesterday from https://github.com/espressif/arduino-esp32

Pablo2048, 2017-11-08:
I'm using websockets in my project on Android device without problem (WIFI_STA mode, little bit older version of core), but this seems like problem with ip addressing to me. Can you please turn on delevoper console on your browser and check, where your application tries to connect to?

nicbkw, 2017-11-08:
Does this help? Chrome browser on my Mac:
[image]

nicbkw, 2017-11-08:
Apologies, hit commit too soon - here is the android phone in remote dev mode on the same Chrome browser:

[image]

Pablo2048, 2017-11-08:
Hmm, but this seems like problem somewhere inside routing or DNS/getHostByName - looks like the phone can't connect to 192.168.1.170 even with ordinary http request...

Pablo2048, 2017-11-08:
Is your MAC connected by wire to the router? (probably you have set AP client isolation?)

nicbkw, 2017-11-08:
Yes, Mac is connected by wire. BT routers provide a BTWiFi connection that is an isolated AP. I have configured an internal WiFi AP, SSID BEAVER on the same network using the same internal DHCP server. The phone and iPad are connected via the BEAVER WiFi AP. The android phone can reach the the router:
[image]

Pablo2048, 2017-11-08:
Yes, but this is the router - not other wifi client. AP isolation means that any wifi station can communicate only with router but can not communicate with other wifi station.

nicbkw, 2017-11-08:
@Pablo2048  Thanks so much for this explanation - I just tried with a python webserver from my home directory on the Mac with python -m SimpleHTTPServer.
http://192.168.1.93:8000 and trying this from my android phone, I see the same error
net::ERR_ADDRESS_UNREACHABLE
All my dev so far has been with servers with a wired network connection, so I didn't see this as an issue. Do you have any clues as to how I disable AP isolation on my private network???

Pablo2048, 2017-11-08:
Ufff, sorry - unfortunately no. Try to search for "AP isolation" on the web - it depend on your AP hardware...

Pablo2048, 2017-11-08:
A small hints:
https://www.howtogeek.com/179089/lock-down-your-wi-fi-network-with-your-routers-wireless-isolation-option/
second hint - try to install Fing on your Android phone and check, if you can see neighbor devices...

nicbkw, 2017-11-08:
Thanks again - I use Fing all the time! The trick on a BT Hub5 seems to be to disable smart setup. The phone can now access the home directory on the Mac:

```
Nics-Mac-mini-2:canvas-gauge nicbkw$ python -m SimpleHTTPServer
Serving HTTP on 0.0.0.0 port 8000 ...
192.168.1.165 - - [08/Nov/2017 12:41:26] "GET / HTTP/1.1" 200 -
192.168.1.165 - - [08/Nov/2017 12:41:27] "GET /style.css HTTP/1.1" 200 -
192.168.1.165 - - [08/Nov/2017 12:41:27] "GET /gauge.min.js HTTP/1.1" 200 -
192.168.1.165 - - [08/Nov/2017 12:41:27] "GET /index.js HTTP/1.1" 200 -
192.168.1.165 - - [08/Nov/2017 12:41:27] code 404, message File not found
192.168.1.165 - - [08/Nov/2017 12:41:27] "GET /favicon.ico HTTP/1.1" 404 -
192.168.1.165 - - [08/Nov/2017 12:41:45] "GET / HTTP/1.1" 200 -

```
and the AsyncWebServer on the heltec-LoRa module

@me-no-dev  - apologies for the confusion

aguaviva, 2019-10-04:
shame

closed 2019-10-04 by aguaviva