me-no-dev/ESPAsyncWebServer #262: ESP32 in STA mode does not serve any Webpage

holema, 2017-10-20:
Im using an ESP32 to connect to an AP.
To serve a Webpage I use this webserver but it dos not serve any webpage.
In AP-mode all works perfectly.
```
WiFi.mode(WIFI_STA);

if (WiFi.getAutoConnect() == true)
{
	WiFi.waitForConnectResult();
	Serial.print("Status:");
	Serial.println(WiFi.status());
	Serial.print("SSID:");
	Serial.println(WiFi.SSID());
	if (WiFi.isConnected()) {
		if (dhcpONOff == 0) {
			WiFi.config(ManuelIP,ManuelGW,ManuelSN);
		}
	}
	if (WiFi.status()== WL_NO_SSID_AVAIL ||WiFi.status() == WL_CONNECT_FAILED)
	{
		
		Serial.println(WiFi.SSID());
	
	}
}else{
	Serial.println("Keine STA Configuriert");
	WiFi.begin("****","*****");
	while (WiFi.status() != WL_CONNECTED) {
	delay(500);
		Serial.print(".");
}
 Serial.println(WiFi.localIP());
}
WiFi.softAPdisconnect();

ws.onEvent(onWSEvent);
setup_webserver(&servera,&ws); // here the handler are attached to the server. in AP this works.
servera.begin();
```

me-no-dev, 2017-10-20:
Rather strange. I generally use the server primarily in sta mode. Do you get at least some http code in the browser? Or nothing at all?

holema, 2017-10-23:
I got nothing at all, but some callbacks were the problem. I`ll refactor them and then try which part is not working.

holema, 2017-11-03:
@me-no-dev Thank for the good support. I was a failure in the connection to the STA not in the Webserver. So after I fix this everything works.

closed 2017-11-03 by holema