me-no-dev/ESPAsyncWebServer #265: Delete AsyncWebServer object causing exception

baoshi, 2017-10-23:
In some applications, it is preferred that the AsyncWebServer object is only created when required, such as WiFiManager type of implementation, the configuration web server is only created if STA cannot connect.
However I found create and delete AsyncWebServer object causes "Fatal exception 9(LoadStoreAlignmentCause)" always.
Here is a simple piece of code:

`#include <ESP8266WiFi.h>
#include <DNSServer.h>
#include <ESPAsyncWebServer.h>
extern "C"
{
#include <user_interface.h>
}

AsyncWebServer *server;

void setup() 
{
  Serial.begin(74880);
  Serial.setDebugOutput(1);
  WiFi.mode(WIFI_AP_STA);
  wifi_station_disconnect();
  server = new AsyncWebServer(80);
  server->begin();
  delay(1000);
  WiFi.mode(WIFI_STA);
  wifi_station_disconnect();
  delay(1000);
  server->reset();
  delete server;
}

void loop() {
  delay(0);
}`

brcddev, 2017-10-31:
in file:ESPAsyncTCP.cpp

void AsyncServer::end(){
  if(_pcb){
  		tcp_arg(_pcb, NULL);
		tcp_recv(_pcb, NULL);
		tcp_err(_pcb, NULL);
		tcp_poll(_pcb, NULL, 0);
		tcp_sent(_pcb, NULL);
		tcp_recved(_pcb, TCP_WND);
		//tcp_close(pcb);

    //cleanup all connections?
    //tcp_abort(_pcb);
    //tcp_arg(_pcb, NULL);
    //tcp_accept(_pcb, NULL);
    if(tcp_close(_pcb) != ERR_OK){
      tcp_abort(_pcb);
    }
    _pcb = NULL;

  }

#if ASYNC_TCP_SSL_ENABLED
  if(_ssl_ctx){
    ssl_ctx_free(_ssl_ctx);
    _ssl_ctx = NULL;
    if(_pending){
      struct pending_pcb * p;
      while(_pending){
        p = _pending;
        _pending = _pending->next;
        if(p->pb){
          pbuf_free(p->pb);
        }
        free(p);
      }
    }
  }
#endif
}

luc-github, 2018-08-05:
@me-no-dev 
I have same issue with ESP32
`E (31125) boot: Assert failed in tcp_abandon, /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/core/tcp.c:375 (don't call tcp_abort/tcp_abandon for listen-pcbs)
`
so I have added `_server.end();` in AsyncWebServer destructor
```
AsyncWebServer::~AsyncWebServer(){
  reset();
  _server.end();
  delete _catchAllHandler;
}
```
and changed abort by close in `AsyncServer::end();`
```
void AsyncServer::end(){
    if(_pcb){
        tcp_arg(_pcb, NULL);
        tcp_accept(_pcb, NULL);
        if(_in_lwip_thread){
            tcp_close(_pcb);
            //tcp_abort(_pcb);
        } else {
            _tcp_close(_pcb);
            //_tcp_abort(_pcb);
        }
        _pcb = NULL;
    }
}
```
seems ok now - is that correct ?

baggior, 2019-04-09:
I confirm the following issue on delete of AsyncWebServer pointer:

`if (asyncWebServer) delete asyncWebServer;  //CRASHES`

assertion "heap != NULL && "free() target pointer is outside heap areas"" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/heap_caps.c", line 268, function: heap_caps_free
abort() was called at PC 0x400e6987 on core 1

Backtrace: 0x4008cad8:0x3ffd4690 0x4008cd09:0x3ffd46b0 0x400e6987:0x3ffd46d0 0x40084ab3:0x3ffd4700 0x40086c19:0x3ffd4720 0x4000bec7:0x3ffd4740 0x401343c9:0x3ffd4760 0x400de4ec:0x3ffd4780 0x40182147:0x3ffd47a0 0x40147b2b:0x3ffd47c0 0x40147e11:0x3ffd47e0 0x40147ea2:0x3ffd4820 0x400dbaab:0x3ffd4840 0x400d88b5:0x3ffd4870 0x400d8963:0x3ffd4890 0x400dc457:0x3ffd48b0 0x401475db:0x3ffd48f0 0x4014509b:0x3ffd4930 0x40145141:0x3ffd4970 0x40145399:0x3ffd49c0 0x40141745:0x3ffd49e0 0x4014178d:0x3ffd4a20 0x40141aa6:0x3ffd4a40 0x4008ed59:0x3ffd4a70

RezaGolriz, 2019-07-02:
Is there fix 

> I confirm the following issue on delete of AsyncWebServer pointer created with new:
> 
> `asyncWebServer = new AsyncWebServer(80);`
> `asyncWebServer -> begin();`
> `if (asyncWebServer) delete asyncWebServer; //CRASHES`
> 
> assertion "heap != NULL && "free() target pointer is outside heap areas"" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/heap_caps.c", line 268, function: heap_caps_free
> abort() was called at PC 0x400e6987 on core 1
> 
> Backtrace: 0x4008cad8:0x3ffd4690 0x4008cd09:0x3ffd46b0 0x400e6987:0x3ffd46d0 0x40084ab3:0x3ffd4700 0x40086c19:0x3ffd4720 0x4000bec7:0x3ffd4740 0x401343c9:0x3ffd4760 0x400de4ec:0x3ffd4780 0x40182147:0x3ffd47a0 0x40147b2b:0x3ffd47c0 0x40147e11:0x3ffd47e0 0x40147ea2:0x3ffd4820 0x400dbaab:0x3ffd4840 0x400d88b5:0x3ffd4870 0x400d8963:0x3ffd4890 0x400dc457:0x3ffd48b0 0x401475db:0x3ffd48f0 0x4014509b:0x3ffd4930 0x40145141:0x3ffd4970 0x40145399:0x3ffd49c0 0x40141745:0x3ffd49e0 0x4014178d:0x3ffd4a20 0x40141aa6:0x3ffd4a40 0x4008ed59:0x3ffd4a70

Is there fix for this? I'm getting this issue and don't know how to handle it!

closed 2019-10-05 by stale[bot]