me-no-dev/ESPAsyncWebServer #268: How to upload a file to SPIFFS large 40K

mcmega, 2017-10-27:
Hi!
In setup server:
```
void onUpload(AsyncWebServerRequest *request, String filename, size_t index, uint8_t *data, size_t len, bool final) {
    if(!index) {
        http_uploading_file = SPIFFS.open(MCU_HEX_PATH, "w");
    }
    for(size_t i = 0; i < len; i++) {
        http_uploading_file.write(data[i]);
    }
    if(final) {
        http_uploading_file.flush();
        http_uploading_file.close();
    }
}

void serverInit(void) {
    server.on("/mcu/upload", HTTP_POST, [](AsyncWebServerRequest *request) {
        request->send(200, "text/plain; charset=utf-8", "OK");
    }, onUpload);
    server.onFileUpload(onUpload);
    server.begin();
}
```
In Web page:
```
onFileChange(event) {
        this.setState({file: event.target.files[0]});
        const {file} = this.state;
        if (file) {
            const data = new FormData;
            data.append('data', file);
            const xhr = new XMLHttpRequest;
            xhr.open('POST', '/mcu/upload');
            xhr.send(data);
        }
    }
```
But I can not transfer a file larger than 40K.
Maybe I'm mis-transmitting from the Web?
Help me to understand.

mcmega, 2017-10-27:
And very long time to upload (4-6 sec) 4k

mcmega, 2017-10-27:
Understood, removed delay(1); from loop fuction.

closed 2017-10-27 by mcmega

GeorgeFlorian, 2019-07-04:
What is `http_uploading_file` ?
Also, can you please post more code that includes uploading files ?
I am also trying to implement upload into my server but I have no idea how.
This is what I've tried but it obviously doesn't work: https://github.com/me-no-dev/ESPAsyncWebServer/issues/542