me-no-dev/ESPAsyncWebServer #275: The app is not loading

mcmega, 2017-10-30:
Hi!
I am use ESP8266 (ESP12), default settings.
The Web interface is very unstable.
3 files are sent to the browser (3 requests):
- index.htm (PROGMEM)
- style.css.gz (SPIFFS)
- main.js.gz (SPIFFS)

I tried to send everything through the server with the addition of the header (avoid CORS errors)
```
// MAIN.JS
server.on("/main", HTTP_GET, [](AsyncWebServerRequest * request) {
    AsyncWebServerResponse *response = request->beginResponse(SPIFFS, "/main.js", "application/javascript; charset=utf-8");
    response->addHeader("Access-Control-Allow-Origin", "*");
    request->send(response);
});

// STYLE.CSS
server.on("/style", HTTP_GET, [](AsyncWebServerRequest * request) {
    AsyncWebServerResponse *response = request->beginResponse(SPIFFS, "/style.css", "text/css; charset=utf-8");
    response->addHeader("Access-Control-Allow-Origin", "*");
    request->send(response);
});
```
But in any way the files do not load. The browser has a status Download, although the size matches. With 5-10 attempts everything is loaded. There are no delays in the sketch. Size main.js.gz 23kB

me-no-dev, 2017-10-30:
Have you tried with full filenames?
```cpp
// MAIN.JS
server.on("/main.js", HTTP_GET, [](AsyncWebServerRequest * request) {
    AsyncWebServerResponse *response = request->beginResponse(SPIFFS, "/main.js", "application/javascript; charset=utf-8");
    response->addHeader("Access-Control-Allow-Origin", "*");
    request->send(response);
});

// STYLE.CSS
server.on("/style.css", HTTP_GET, [](AsyncWebServerRequest * request) {
    AsyncWebServerResponse *response = request->beginResponse(SPIFFS, "/style.css", "text/css; charset=utf-8");
    response->addHeader("Access-Control-Allow-Origin", "*");
    request->send(response);
});
```

mcmega, 2017-10-30:
Yes, it's getting better, but sometimes there are problems. The status is 200, but there is no content.

mcmega, 2017-10-31:
I noticed that after 3-5 minutes the server stopped responding and the pages did not load.

rick-bp, 2017-11-01:
Hi,
Let me try help a bit.
Why you dont use: server.serveStatic("/", SPIFFS, "/"); to serve all files inside filesystem?
TIP one: if you have many files to load on the html page, I sugest you embebed all in one file and load this one.
TIP two: Embebed all image files on html page as base64. I use this site to do this: https://www.base64-image.de

I hope had help you! good luck!

mcmega, 2017-11-01:
Thank you!
I can not return files directly from the file system because errors occur CORS. Therefore, I add a header to the server
```response->addHeader("Access-Control-Allow-Origin", "*");```
There is a sense favicon.ico as bass64 add to the index page (PROGMEM), or by reference?
<link rel="shortcut icon" type="image/x-icon" href="favicon.ico"/>

mcmega, 2017-11-03:
I figured out, the problem was that I used ESPAsyncWebServer and Client esp8266/Arduino in different places of code. You need to switch to AsyncClient (this library)!

closed 2017-11-03 by mcmega

rick-bp, 2017-11-04:
mcmega, I embebed the images files on HTML file.
like this:
[image]

rick-bp, 2017-11-04:
`<img src="data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAABgAAAAYCAYAAADgdz34AAAABmJLR0QA/wD/AP+gvaeTAAAAzElEQVRIie2SMQrCQBBF34hYiKAiFla2HsHOW3iH9BapU0s6L2DrRbyLiAoKIsLYTCFE5LNE0vi6fezwB+bDjzH1o7vPgV48r2a2r20Ld8+8SqbMtsSMmeiSA5JRA56iq9AWA7bAAhjG+xSueaSaunsHWALdUDdgZ2aPWrZw9/xDTXNlVj3yRHTJAcmoAXfRVVBrugGmwCjex3DNo9Z0AKyAcagDsDazcy1buHvxoaaFMqseuS+65IBk1ICL6CqoNS1jmfcjl+Lsn++8AIlCYvpJCZHQAAAAAElFTkSuQmCC" width="24" height="24" class="d-inline-block align-middle">`