me-no-dev/ESPAsyncWebServer #278: JSON Parameters in HTTP_POST Request

wahabkhan1892, 2017-11-02:
I am working on a project in which the ESP8266 starts in softAP mode on start up. The homepage displays a list of available WiFi networks. The user selects one of them and enters the password for that network. This sends a HTTP_POST request to the web server which includes the ssid, and password of the selected WiFi network a JSON list. But when I check the number of parameters for the request, I'm getting 0. The server identifies the content type and content length correctly, but it can't recognize any parameters. Here is the code for the "/connect" handle. I have also tried using `request->args()`. That doesn't work either.

```
    server.on("/connect", HTTP_POST, [](AsyncWebServerRequest *request)
    {
        String argList;
        String password;
        String key = "ssid";
        Serial.println(request->contentType());
        Serial.println(request->contentLength());
        Serial.print("ArgNum: ");
        Serial.println(request->params());
        for (int i = 0; i < request->params(); i++)
        {
            AsyncWebParameter* p = request->getParam(i);
            Serial.print(p->name());
            Serial.print(": ");
            Serial.println(p->value());
        }

        request->send(200, "text/plain", "SUCCESS");
    });
```

mcmega, 2017-11-02:
You need a parser to handle json. Pay attention to the library https://bblanchon.github.io/ArduinoJson/

mcmega, 2017-11-03:
Вечером скину код, как у меня реализовано.

wahabkhan1892, 2017-11-03:
How is the synchronous webserver able to get the parameters without any JSON parser?

me-no-dev, 2017-11-03:
it does not. maybe post your "POST" javascript code ;) did you check if it does not come in onBody?

wahabkhan1892, 2017-11-03:
How do I access request body data? I want to pass it to JSON parser.

Pablo2048, 2017-11-03:
??? did You read the doc? https://github.com/me-no-dev/ESPAsyncWebServer#body-data-handling

wahabkhan1892, 2017-11-03:
Yes I did, but this function is not being called. I inserted some debug code into `handleBody` and its only being called when `onNotFound` handle is active, and not from any other handles. So I'm confused, how do I access body data for parsing?

Pablo2048, 2017-11-03:
It's one of parameters for .on() handler - I'm using it this way:
  www.on("/saverulesxml", HTTP_POST, www_XMLrules, NULL, www_bodyXMLrules);
I don't remember where I got this info/knowledge, but I think it was somewhere in doc or some header...

wahabkhan1892, 2017-11-03:
Thanks, that did the trick.

closed 2017-11-03 by wahabkhan1892