me-no-dev/ESPAsyncWebServer #279: Reboot after 3 failed logins on upload.

CelliesProjects, 2017-11-03:
I am testing some upload code but got reboots of the esp with the following code after 2 or 3 login attempts with the wrong password.

Files are uploaded just fine with:
```bash
cellie@cellie-Mint-64 ~ $ curl --user admin:admin -F 'data=@/home/cellie/Desktop/161204_Mastering_the_FreeRTOS_Real_Time_Kernel-A_Hands-On_Tutorial_Guide.pdf'  '192.168.2.3/api/upload'
```
But when I use the wrong password `admi` instead of `admin` the box reboots after 2 or 3 attempts with this tracelog:
```c
assertion "new_rcv_ann_wnd <= 0xffff" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/core/tcp.c", line 629, function: tcp_update_rcv_ann_wnd
abort() was called at PC 0x400d88ab on core 1

Decoding 8 results
0x400d88ab: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c line 63 (discriminator 8)
0x40087b30: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 553
0x40087c2f: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 553
0x400d88ab: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c line 63 (discriminator 8)
0x400eaa92: tcp_update_rcv_ann_wnd at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/core/tcp.c line 834
0x400eab08: tcp_recved at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/core/tcp.c line 834
0x400d2729: _tcp_recved_api(tcpip_api_call*) at /home/cellie/Arduino/hardware/espressif/esp32/libraries/AsyncTCP/src/AsyncTCP.cpp line 660
0x400e8f11: tcpip_thread at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/api/tcpip.c line 474
```

The offending code:
```c
/****************************************************************************

       Upload a file to esp32 with authorization

       Use curl to upload a file:

       curl --user admin:admin -F 'data=@FILENAME' 'ESP32_IP/api/upload'

 *****************************************************************************/
#include "FS.h"
#include "SD.h"
#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>

AsyncWebServer server(80);

const char* ssid = "huiskamer";
const char* password = "0987654321";
const char* http_username = "admin";
const char* http_password = "admin";
 
void setup() {
  Serial.begin( 115200 );

  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  Serial.printf( "Connecting to SSID %s with PSK %s...\n", ssid, password  );
  if (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.printf("WiFi Failed!\n");
    return;
  }
  Serial.println( WiFi.localIP() );
  
  server.on( "/api/upload", HTTP_POST, [](AsyncWebServerRequest * request)
  {
    request->send(200);
  },
  []( AsyncWebServerRequest * request, String filename, size_t index, uint8_t *data, size_t len, bool final )
  {
    static bool   _authenticated;
    static time_t startTimer;

    if ( !index )
    {
      _authenticated = false;
      if ( request->authenticate(http_username, http_password) )
      {
        startTimer = millis();
        Serial.printf( "UPLOAD: Started to receive '%s'.\n", filename.c_str() );
        _authenticated = true;
      }
      else
      {
        Serial.println( "Unauthorized access." );
        return request->send( 401, "text/plain", "Not logged in." );
      }      
    }

    if ( _authenticated )
    {
      //Serial.printf( "%i bytes received.\n", index );
      //Store or do something with the data...
    }
    else
    {
      Serial.println( "Unauthorized access." );
      return request->send( 401, "text/plain", "Not logged in." );
    }  

    if ( final && _authenticated )
    {
      Serial.printf( "UPLOAD: Done. Received %.2f kBytes in %.2fs which is %i kB/s.\n", index / 1024.0, ( millis() - startTimer ) / 1000.0, index / ( millis() - startTimer ) );
    }
  });

  server.onNotFound( []( AsyncWebServerRequest * request )
  {
    Serial.printf("NOT_FOUND: ");
    if (request->method() == HTTP_GET)
      Serial.printf("GET");
    else if (request->method() == HTTP_POST)
      Serial.printf("POST");
    else if (request->method() == HTTP_DELETE)
      Serial.printf("DELETE");
    else if (request->method() == HTTP_PUT)
      Serial.printf("PUT");
    else if (request->method() == HTTP_PATCH)
      Serial.printf("PATCH");
    else if (request->method() == HTTP_HEAD)
      Serial.printf("HEAD");
    else if (request->method() == HTTP_OPTIONS)
      Serial.printf("OPTIONS");
    else
      Serial.printf("UNKNOWN");
    Serial.printf(" http://%s%s\n", request->host().c_str(), request->url().c_str());
    request->send( 404, "text/plain", "Not found." );
  });

  server.begin();

  Serial.println( "Ready for upload" );
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

CelliesProjects, 2017-11-05:
Seems I have it working stable now by replacing: 
```c
    if ( _authenticated )
    {
      //Serial.printf( "%i bytes received.\n", index );
      //Store or do something with the data...
    }
    else
    {
      Serial.println( "Unauthorized access." );
      return request->send( 401, "text/plain", "Not logged in." );
    }  
```
by
```c
    if ( _authenticated )
    {
      //Serial.printf( "%i bytes received.\n", index );
      //Store or do something with the data...
    }
```
So only return a 401 on the first request and ignore the rest of the request.

Also replaced 
```c
  server.on( "/api/upload", HTTP_POST, [](AsyncWebServerRequest * request)
  {
    request->send(200);
  },
```
with:
```c
  server.on( "/api/upload", HTTP_POST, []( AsyncWebServerRequest * request )
  {
    if ( request->authenticate( www_username, www_password ) )
      {
        request->send( 200 );
      }
      else
      {
        request->requestAuthentication();
      }
```
and use that as a pre-flight check to see if the client is logged in.

See the full example at [minimalUploadAuthESP32](https://github.com/CelliesProjects/minimalUploadAuthESP32)

The example above works if modified to:
```c
/****************************************************************************

       Upload a file to esp32 with authorization

       Use curl to upload a file:

       curl --user admin:admin -F 'data=@FILENAME' 'ESP32_IP/api/upload'

 *****************************************************************************/
#include "FS.h"
#include "SD.h"
#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>

AsyncWebServer server(80);

const char* ssid = "--------";
const char* password = "--------";
const char* http_username = "admin";
const char* http_password = "admin";
 
void setup() {
  Serial.begin( 115200 );

  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  Serial.printf( "Connecting to SSID %s with PSK %s...\n", ssid, password  );
  if (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.printf("WiFi Failed!\n");
    return;
  }
  Serial.println( WiFi.localIP() );
  server.on( "/api/upload", HTTP_POST, []( AsyncWebServerRequest * request )
  {
    if ( request->authenticate( www_username, www_password ) )
    {
      request->send( 200 );
    }
    else
    {
      request->requestAuthentication();
    },
  []( AsyncWebServerRequest * request, String filename, size_t index, uint8_t *data, size_t len, bool final )
  {
    static bool   _authenticated;
    static time_t startTimer;

    if ( !index )
    {
      _authenticated = false;
      if ( request->authenticate(http_username, http_password) )
      {
        startTimer = millis();
        Serial.printf( "UPLOAD: Started to receive '%s'.\n", filename.c_str() );
        _authenticated = true;
      }
      else
      {
        Serial.println( "Unauthorized access." );
        return request->send( 401, "text/plain", "Not logged in." );
      }      
    }

    if ( _authenticated )
    {
      //Serial.printf( "%i bytes received.\n", index );
      //Store or do something with the data...
    }

    if ( final && _authenticated )
    {
      Serial.printf( "UPLOAD: Done. Received %.2f kBytes in %.2fs which is %i kB/s.\n", index / 1024.0, ( millis() - startTimer ) / 1000.0, index / ( millis() - startTimer ) );
    }
  });

  server.onNotFound( []( AsyncWebServerRequest * request )
  {
    Serial.printf("NOT_FOUND: ");
    if (request->method() == HTTP_GET)
      Serial.printf("GET");
    else if (request->method() == HTTP_POST)
      Serial.printf("POST");
    else if (request->method() == HTTP_DELETE)
      Serial.printf("DELETE");
    else if (request->method() == HTTP_PUT)
      Serial.printf("PUT");
    else if (request->method() == HTTP_PATCH)
      Serial.printf("PATCH");
    else if (request->method() == HTTP_HEAD)
      Serial.printf("HEAD");
    else if (request->method() == HTTP_OPTIONS)
      Serial.printf("OPTIONS");
    else
      Serial.printf("UNKNOWN");
    Serial.printf(" http://%s%s\n", request->host().c_str(), request->url().c_str());
    request->send( 404, "text/plain", "Not found." );
  });

  server.begin();

  Serial.println( "Ready for upload" );
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

closed 2017-11-20 by CelliesProjects

acris5, 2018-08-31:
Thank for the recipe, also worked for me
I removed request->send(200); from request body handler  and now received all parts of multipart request.