me-no-dev/ESPAsyncWebServer #281: Please release point version

andig, 2017-11-08:
.. this will allow everybody to use the updated `library.json` files with staging version compatibility.

*THANKS*

closed 2019-10-05 by stale[bot]

andig, 2019-10-05:
Its kinda unsatisfying when every single issue is closed by the stale bot without any fuether action.

me-no-dev, 2019-10-06:
but you did not react in the two weeks that the bot notified you for :) the bot is here because there are many old issues. You acting on an issue that the bot has marked, notifies me that this is still relevant and goes to separate inbox.

andig, 2019-10-06:
I see. Didn‘t realize I could convonce the bot its still relevant. But then- id something as simple as a release didn‘t happen in 2 years- why should it now 😂

me-no-dev, 2019-10-06:
it's WIP :P not up to release