me-no-dev/ESPAsyncWebServer #282: Page never stops loading, content-length one short?

JamesNewton, 2017-11-08:
I'm seeing an interesting problem. Using ESP-8266 Core 2.3.0 and ESPAsyncWebServer from mid october (I will upgrade to latest after I document this and test again). I'm using ESPAsyncTCP from a prior release as well to avoid the issue with labels changing in Core 2.4.0. I generally don't like to be on the bleeding edge.

Anyway, Chrome and IE both seemed to not want to display the web pages... they would connect quickly, the debug print from the device indicated it had served the page, but the icon just kept spinning and nothing was displayed. The network devtool page in chrome would just show "loading".  After a long while (minutes) the page would suddenly show up, and from then on, it would serve without issue. When it would work, the right click / Copy / Copy all as HAR data would show a content-length of 1306 in the header, and the same length for the body. When it doesn't work, after the time out, it showed 1306 for the content-length and 1305 for the body.

So I found a little utility called WebBug that I had in the "box o tricks" from years back which does a very simple http 0.9, 1.0, or 1.1 GET and shows the exact response. That /always/ shows:

````
HTTP/1.1 200 OK
Content-Length: 1306
Content-Type: text/html
Connection: close
Cache-Control: max-age=86400
Accept-Ranges: none
````
With the actual content below it, but if I copy out the actual content returned on the line below, and select it, my editor always shows 1305 bytes selected... AND... the status on the little utility shows "Receiving"... forever. With other web sites, the content-length and actual content match and the status goes right back to "Disconnected" 

I've checked the actual content and it is 1305 bytes. This is a bit unsure as the Arduino IDE has both carriage return and linefeeds and the served content does not have the carriage returns. So to do a correct comparison I'm loading the source content into 101 Editor (a hex capable editor) and replacing all the 0D's with nothing. At that point, it perfectly matches the served content and has a length of 1305. 

So... I /think/ that the content-length is being misreported by 1, which keeps the connection open and in some cases, not sure why not all cases, causes the browser to hang waiting for that last byte before processing the page. It really bugs me that the browser works /sometimes/ and not others. I want to trust Chrome a lot more than this little WebBug utility. I /think/ Chrome is compensating automatically in some cases, but I can't figure out why or how. 

I don't think my current code would be worth publishing because there is a lot of stuff in there which is unrelated, but if you want it, just ask. In general I'm serving 
`const char PAGE_AdminMainPage[] PROGMEM = R"=====(`
(1305 bytes)
`)=====";`
via a very basic:
`AsyncWebServer server(80);`

and then in the setup()
````
    server.on ( "/admin.html", HTTP_GET, [](AsyncWebServerRequest *request) { //settings menu
#ifdef DEBUGGING
	  debugbuf += "admin.html";
#endif
	  AsyncWebServerResponse *response = request->beginResponse(200, "text/html", PAGE_AdminMainPage); 
	    response->addHeader ( "Cache-Control", "max-age=86400" );  //doesn't change, let the browser know
	  request->send ( response );   
	  }  );
````
The content is NOT being served from the cache by Chrome despite the extra header, and it seems to have the same issue no matter if that is there or not. 

Next steps:
1. Upgrade to the latest version of ESPAsyncWebServer

2. Install the reference sample .ino from 
https://github.com/me-no-dev/ESPAsyncWebServer#setting-up-the-server
and see if the issue persists. If not, try to figure out the difference between that and my code. 
3. Tweek the code to return one byte less for content-length and see if that fixes the issue.

And any other ideas for troubleshooting would be much appreciated.

me-no-dev, 2017-11-08:
try with: ```request->beginResponse(200, "text/html", (const uint8_t *)PAGE_AdminMainPage, strlen(PAGE_AdminMainPage));```, or ```request->beginResponse_P(200, "text/html", PAGE_AdminMainPage);```

me-no-dev, 2017-11-09:
can you please add somewhere ```Serial.printf("Response Length: %u\n", PAGE_AdminMainPage);``` so we can see where that extra byte comes from?

JamesNewton, 2017-11-10:
Response Length:1306

I guess it counts the second line feed after the response header? Now that I look at the source view in Chrome, it does show a blank line at the top. That's probably where I'm getting off by one.

No idea why WebBug is showing the connection remains open. Don't really care.

I'm noticing that when I have the problem, if I just reprogram the chip, it suddenly works perfectly... I've reduced the upload speed baud rate and that doesn't seem to have any effect. I've also tried a different ESP, same issues. I can program it 4 or 5 times in a row and it works each time, the make changes, program, fail, program, works. 

Anyway, there is nothing wrong in AsyncWebServer so I'm closing this and THANK YOU very much for your help.

closed 2017-11-10 by JamesNewton