me-no-dev/ESPAsyncWebServer #283: ESPAsyncWebClient ??

timpur, 2017-11-13:
Wondering if there is already some where an `ESPAsyncWebClient` or should there be ? Mainly to replace the built in `HTTPClient` of `esp8266` but uses the [ESPAsyncTCP](https://github.com/me-no-dev/ESPAsyncTCP). 

Also should the `ESPAsyncWebServer` use this? 

Also this WebClient would also support the `esp32` since there is a [AsyncTCP ](https://github.com/me-no-dev/AsyncTCP) lib for it now thanks to you (@me-no-dev)

judge2005, 2017-12-05:
https://github.com/judge2005/ESPAsyncHttpClient

timpur, 2018-01-15:
@me-no-dev any plans on this?

timpur, 2018-02-12:
Thinking of create a replacement for https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.h but using asyncTCP lib

closed 2019-10-05 by stale[bot]