me-no-dev/ESPAsyncWebServer #288: Why send back response before executing handleUpload

kdobrev, 2017-11-19:
Hi, I made a program based on your example and my sketch rends the response before execution of handleUpload. I find this inconvenient since I may have not uploaded anything.
Here is my code:
```
  server.on("/admin.html", HTTP_POST, [](AsyncWebServerRequest * request) {
    //    Serial.println("/admin.html post");
    //    handleUpload();
    //    request->send(SPIFFS, "/admin.html");
    //  });
    AsyncWebServerResponse *response = request->beginResponse(200, "text/plain", "OK");
    response->addHeader("Connection", "close");
    request->send(response);
    //request->send(SPIFFS, "/admin.html?status=uploaded");
  }, handleUpload);
```

```
void handleUpload(AsyncWebServerRequest *request, String filename, size_t index, uint8_t *data, size_t len, bool final) {
  if (!index) {
    Serial.printf("UploadStart: %s\n", filename.c_str());
    if (!filename.startsWith("/"))
      filename = "/" + filename;
    if (SPIFFS.exists(filename)) {
      SPIFFS.remove(filename);
    }
    uploadFile = SPIFFS.open(filename, "w");
    //filename = String();
  }
  for (size_t i = 0; i < len; i++) {
    uploadFile.write(data[i]);
  }
  if (final) {
    uploadFile.close();
    //    uploadFile = SPIFFS.open(filename, "r");
    //    if(!uploadFile) Serial.printf("Greshka: %s, %u B\n", filename.c_str(), index + len);
    //    Serial.printf("UploadEnd: %s, %u B\n", filename.c_str(), index + len);
    outbound_transfer(filename);
  }
}

```
And I have another issue with  outbound_transfer(filename);  - it give me Soft WDT reset. Do I need to just raise a flag and put it in main loop?

```
void outbound_transfer(String filename) {
  filename = "/style.css";
  File f = SPIFFS.open(filename, "r");
  size_t fsize = f.size();
  const byte numChars = 256;
  char receivedChars[numChars]; // an array to store the received data
  char rc;
  
  while (f.available() > 0) {
    for (int i = 0; i <= 255; i++) {
      rc = f.read();
      Serial.print(rc);
    }
    while (!Serial.available() > 0) {
      //Serial.println("Waiting of 'OK!'");
    }
    for (static byte i = 0; Serial.available() > 0 || i < 4; i++) {
      rc = Serial.read();
    }
  }
  f.close();
}

```

me-no-dev, 2017-11-19:
@kdobrev българин? Is this ESP8266? Here is what I will propose to you ;)
first do not use ```uploadFile```. Use ```request->_tempFile``` instead. That way each request can have it's own file and you will not get things mixed up if another call comes in.

Then do not print out the file at the end of ```onUpload``` but instead do it after ```request->send(response);``` in the regular handler. You can get the filename from the POST variables (see README for info on how to get just the file), then open it and print. If that also triggers WDT, maybe your Serial speed is too low (try 115200 or more). Another option is to save the filename and a flag and do that in the loop.

```SPIFFS.open(filename, "w");``` will truncate the file anyway so you do not need to delete it if it's already there (that will save you time).

kdobrev, 2017-11-19:
@me-no-dev 
Да, булгар! :)
My serial speed is 9600 and the remote MCU I'm sending the data is writing to its EEPROM so total time will be around 1 min for this 16kB file.
Can I spend so much time in my ` server.on("/admin.html", HTTP_POST, [](AsyncWebServerRequest * request)` ?

me-no-dev, 2017-11-19:
oh no, that is too long time. 50-100ms is about as much as you can and even that is not recommended at all (you are blocking the network thread). Също булгар ;)

kdobrev, 2017-11-22:
Hi, again,
I am trying and succeeded so far in what you suggested.
I am using request->_tempFile and outbound_transfer() is in main loop(). 
Now I want to reuse server.on for different POST requests. In other words if I have different forms that call the same admin.html and some upload a file and other just send a command. How do I provide the `uint8_t *data, size_t len` to handleUpload(), when inside server.on();

```
`  server.on("/admin.html", HTTP_POST, [](AsyncWebServerRequest * request) {
    String rtcode;

    //upload file
    if (request->hasParam("file", true, true)) {
      rtcode = "OK";
      //handleUpload();
      handleUpload(request, filename, size_t index, uint8_t *data, size_t len, bool final);
    }
    
    //do not upload file, do something else
    else if (request->hasParam("prepare_file", true)) {
      rtcode = "PREP";
      inbound_transfer = 1;
    }
    
    //do nothing
    else {
      rtcode = "FAIL";
    }
    
    AsyncWebServerResponse *response = request->beginResponse(200, "text/plain", rtcode);
    response->addHeader("Connection", "close");
    request->send(response);
  //}, handleUpload);
  });`
```

me-no-dev, 2017-11-23:
you can not do that ;) the first callback is called AFTER all uploads and body parsing is done. You need to use the proper handler

closed 2017-11-23 by me-no-dev