me-no-dev/ESPAsyncWebServer #289: AsyncWebServer crashes when Read file in SPIFFS

HugoSH, 2017-11-21:
Hardware: NodeMCU
Software: Arduino 1.8.5 + Arduino for ESP8266 2.3.0

Hope this is not a re-post. Otherwise, please advice where to look for solution. 
Thanks!

I am using the AsyncWebServer and SPIFFS.
Write to SPIFFS works ok.
But when Read a file from SPIFFS, it crashes.
Please advice.  Thanks !

Error message from Serial output: 
Panic /Users/hugo/Library/Arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/core_esp8266_main.cpp:98 __yield

ctx: sys
sp: 3ffffb60 end: 3fffffb0 offset: 01b0

stack>>>
3ffffd10: 3fff194c 00000001 3ffffd78 4020fc5d
3ffffd20: 0000000a 3ffffd60 3ffffd78 4020e8ad
3ffffd30: 3ffe0029 00000049 3ffffd60 4020dd11
...
<<<stack<<<
ets Jan 8 2013,rst cause:2, boot mode:(1,6)

ets Jan 8 2013,rst cause:4, boot mode:(1,6)

wdt reset

Code is as follows;
`
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <ESPAsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <FS.h>
#include <time.h>

AsyncWebServer server(80);

void indexLoad(AsyncWebServerRequest *request) ;
void indexSubmit(AsyncWebServerRequest *request) ;

const char* ssid = "ChinaNet-tz4i";
const char* password = "qfk9n6en";
const char * hostName = "ESP-async";

void setup() {
Serial.begin(115200);
Serial.println(" Async WebServer Test ");
Serial.setDebugOutput(true);
WiFi.hostname(hostName);
WiFi.mode(WIFI_AP_STA);
WiFi.softAP(hostName);
WiFi.begin(ssid, password);
if (WiFi.waitForConnectResult() != WL_CONNECTED) {
delay(500);
WiFi.begin(ssid, password);
}
Serial.print("STA mode IP address:");
Serial.println(WiFi.localIP());
if(SPIFFS.begin()){
Serial.println("SPIFFS mount success.");
server.serveStatic("/scripts", SPIFFS, "/scripts", "max-age=86400");
}
else Serial.println("SPIFFS mount failure.");

MDNS.addService("http","tcp",80);

server.on("/", HTTP_GET, indexLoad);
server.on("/index", HTTP_GET, indexLoad);
server.on("/index", HTTP_POST, indexSubmit);
server.begin();
}

void loop() {

}

void indexLoad(AsyncWebServerRequest *request) {
Serial.println("\nShow index.html ");
request->send(SPIFFS, "/index.html");
}

void indexSubmit(AsyncWebServerRequest *request) {
String inputValue, tempStr;

  Serial.println("index Page Submit()");
  inputValue = request->arg("inputTxt");
  File registerFile = SPIFFS.open("/scripts/registration.js", "r");   //crashes here
  if(!registerFile){
    Serial.println("Failed to Read /scripts/registration.js ");
  }
  else{
            while(registerFile.available()){
                tempStr += registerFile.readStringUntil('\n');
            }
            registerFile.close();    
            Serial.println(tempStr);
  }       
  Serial.println("index page submit() End");     
}

me-no-dev, 2017-11-21:
Issue is in your code ;) you are calling ```registerFile.readStringUntil('\n');``` which is calling yield() and that is not allowed in Async ;)

HugoSH, 2017-11-21:
Thanks so much!
Let me change to read one byte at a time.

HugoSH, 2017-11-21:
by switching to : char(registerFile.read());
it appears the crash is avoided. 
I will do more tests.

pranjal-joshi, 2020-06-03:
Hello @me-no-dev,

I am trying to read a file from SPIFFS inside AsyncWebRequestCallback as follow:
(Part of code is given instead of entire code!)
```
if(SPIFFS.begin()) {
    uint16_t i;
    char buf[JSON_BUF_SIZE];
    File f = SPIFFS.open(fn,"r");
    if(!f) {
      if(DEBUG)
         printDebug("ERROR: Failed to open file for reading.");
         return FileOpenError;
    }
    f.readBytes(buf,f.size());
    f.close();
    deserializeJson(*doc, buf);
}
```
Please let me know the correct way to this without getting a memory dump (ESP wdt reset).