me-no-dev/ESPAsyncWebServer #290: WiFi_STA does not re-connect when AsynWebServer is running

HugoSH, 2017-11-22:
Device: NodeMCU 1.0
IDE: Arduino 1.8.5

This is a must-do step in wifi configuration. But I stumbled in step 2). Thanks for any advice. 
1)After ESP stars as AP, AsynWebServer provides index page to collect Router SSID/PSK.  (OK)
2)Then ESP tries to connect to Router with acquired credentials. (Never get connected with AsynWebServer is running. Please advice .... Thanks!)
3)If fails, ESP reverse back to AP mode. 

`
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <ESPAsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <FS.h>  

AsyncWebServer server(80);

const char* ssid = "myRouterSSID";
const char* password = "myRouterPSK";
const char * hostName = "esp-Async";

void indexLoad(AsyncWebServerRequest *request) ;
void indexSubmit(AsyncWebServerRequest *request) ;

void setup(){
  Serial.begin(115200);
  Serial.println("\n Async WebServer Lab"); 
  Serial.setDebugOutput(false);
  WiFi.hostname(hostName);
  WiFi.mode(WIFI_AP_STA);
  WiFi.softAP(hostName);
  WiFi.begin(ssid, password);
  if (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.printf("STA: Failed!\n");
    WiFi.disconnect(false);
    delay(500);
    WiFi.begin(ssid, password);
  }
  Serial.print("STA mode IP address:"); 
  Serial.println(WiFi.localIP());
  if(SPIFFS.begin())
           Serial.println("SPIFFS mount success."); 
  else  Serial.println("SPIFFS mount failure.");   
  MDNS.addService("http","tcp",80);
  server.on("/", HTTP_GET, indexLoad);
  server.on("/index", HTTP_GET, indexLoad);
  server.on("/index", HTTP_POST, indexSubmit);
  server.begin();

void loop() {
 // nothing here. it's AsyncWebServer :)
}

void indexLoad(AsyncWebServerRequest *request) {
      Serial.println("\nShow index.html ");
      request->send(SPIFFS, "/index.html");
}

void indexSubmit(AsyncWebServerRequest *request) {           
      String inputSSID, inputPassword;
      Serial.println("indexSubmit()");
      inputSSID = request->arg("txtSSID");  
      inputPassword = request->arg("txtPSK");
      Serial.println("Attempt to connect to Router.");
      WiFi.disconnect();
      WiFi.mode(WIFI_STA);                         
      Serial.println("Change mode to STA. Connecting to user defined Router...\n");
      WiFi.begin(inputSSID, inputPassword);
      unsigned long timeout = millis();
      while (WiFi.status()!=WL_CONNECTED){              
                ESP.wdtFeed();
                 if (millis() - timeout > 10000 ){          //extend this time to 30s, still not get connection
                      Serial.println("\nError: Failed to connect to Router.");    
                      break;            
                 }
        }           
        if (WiFi.status()==WL_CONNECTED) {          
                    Serial.println("\nConnect to router Wi-Fi: Succss");
        }
      Serial.println("indexSubmit() End");     
}
`

Followup question would be: 
How to restart the AsyncWebServer again? 
Shall I go through the setup() process? 
Thanks!

HugoSH, 2017-11-23:
By setting 
'''arduino
Serial.setDebugOutput(true); 
'''
I noticed that it is actually connected. However, I am still puzzled by the failure in the waiting for connection while loop.

me-no-dev, 2017-11-23:
Ahh... it's a really bad idea to do such a while loop in an async callback ;) Since you are connected over AP and that is how you send the data to the ESP, but then you are setting the mode to STA only, I imagine many things are going wrong and you are not getting the reply back (you effectively stopped AP from working)

HugoSH, 2017-11-23:
Happy Thanksgiving! :)
Much appreciate you looking into my questions. 
What is the proper way to handle this AP to STA mode change in async style?
Or, I do not make the mode change, just call WiFi.begin() for STA mode connection.

me-no-dev, 2017-11-23:
change mode to WIFI_AP_STA to keep the AP on, or set a flag and then do the connect in the loop, so the request is actually closed. It is a good question though... i'll think if i can add something that would allow what you want

HugoSH, 2017-11-25:
Got it. Thanks!
so, not to change the connection mode, maintain it as AP_STA, and establish connection in loop(). 
Is it possible to Stop AsyncWebServer, do the Wi-Fi connection and then ReStart AsyncWebServer? 
Thanks again.

me-no-dev, 2017-11-25:
you do not really need to do that :) server will keep running on the new interface (STA)

HugoSH, 2017-11-28:
By setting a flag in the void indexSubmit(AsyncWebServerRequest *request); 
and then check the flag in the loop() and perform WiFi.status() check in the loop(). 
The issue is now resolved. 
Thanks! Also, thanks for @igrr advice as well.

closed 2017-11-28 by HugoSH

stefanache, 2019-07-21:
what is final source?