me-no-dev/ESPAsyncWebServer #294: ESPAsyncWebSever becomes unresponsive.

bill-orange, 2017-11-27:
I am compiling on the latest released ESP8266 IDE using a WEMOS D1 mini Pro.  ESPAsyncWebSever is included in the FSBrowserNG project that I am using to acquire network and Wundergound credentials.  Everything that is supposed to work works and very fast!  

The problem is that the web server becomes unresponsive after a while.  I have found several posts on this problem dating back years and I found several proposed solutions.

The time is somewhat variable but it consistently becomes unresponsive overnight. I have tried several solutions.  I tried doing a forcearp() every second. It seemed to help.  I tried adding a ping (running in loop) to itself every few minutes.  That also helped but neither possible solution resolved the problem entirely. 

I am nearly out of ideas.  I understand from other posts that if four connections remain open (did I get that right?), it intentionally becomes unresponsive.  Perhaps I could add a reset if this counter gets to 4.  That seems rather drastic and I don't know where to look in the code for this counter.

Thought?  Suggestions? Sympathy?

me-no-dev, 2017-11-28:
this is rather strange... could maybe have a request that is unhandled?

bill-orange, 2017-11-28:
Regarding the possibility of unhandled requests, how can I tell if that is the problems?  It would surprise me if that was the problem since the same web pages accessed by the same workstations work fine with the non-async web server.

me-no-dev, 2017-11-28:
I can also say that there are many instances of running Async without issues :)
One option to have unhandled request is to have a handler and case where you do not send a response (cause of some ```if``` case or similar). Honestly I can not come up with another now :D 

Maybe show some Async code?

bill-orange, 2017-11-28:
"One option to have unhandled request is to have a handler and case where you do not send a response"  
Can you show me sample code for this approach?  Is there a way from within loop() to properly detect if the web browser has become unresponsive and reset it?  Here's my code:

https://1drv.ms/f/s!AmXKqAwyCrbxh9wKhqi1tWbidmsKow

You will notice the PING (including a forcearp() call)  in loop().  That does not solve the problem, but it seems to help.

Here's what the result looks like in its non-async form:

http://orangecaweather.duckdns.org:8021/

I should also mention since it might be a clue, that the async web browser will not work with Safari on my iPhone.  I can open the graph once or twice but after that it hangs up.  To show the graph again in Safari, it is necessary to close Safari with an upward swipe.  Chrome works fine.

me-no-dev, 2017-11-28:
that seems to be the regular web server code though?

bill-orange, 2017-11-28:
The code in the one-drive link is the async version.  FSBrowser uses Async and is very fast In generating the web pages in DATA.

The code at http://orangecaweather.duckdns.org:8021/ is the non-async version.  I have not exposed the async version with an external link because my credential collected in that version are not very secure.  I just provided this link to give you an idea of what I am trying to do.  FSBrowser lets me collect credential without having to compile with them included.

me-no-dev, 2017-11-28:
where is that async code? I do not see it?

bill-orange, 2017-11-28:
It is in my #include list and is initialized and run by FSWebServerLib.  A worthy project indeed.  It is here:

https://github.com/gmag11/FSBrowserNG/blob/master/src/FSWebServerLib.h

me-no-dev, 2017-11-28:
Have to say, this lib is pretty big and pretty old in terms of code. Have you submitted an issue there?

bill-orange, 2017-11-28:
Yes, but no reply. 

Is there is a way to discover, within loop that the async web server has stopped responding?  If so, I can just reset. That approach avoids solving the problem but it is good enough for me!

bill-orange, 2017-12-01:
This issue is certainly vexing.  I have reduced the non-responsive behavior by pinging myself but I don't think it will eliminate the problem.  There is an ongoing discussion here:

https://github.com/esp8266/Arduino/issues/2330

I can add one new observation.  Connecting with Safari on my iPhone seems to induce the unresponsive behavior  rather quickly.

klaasdc, 2018-06-13:
Old topic, but I have the same problem. I use an ESP in a smart home system. It publishes some states to an MQTT server, but also serves webpages (parts from SPIFFS) to be able to configure it. After a while (could be few hours, to few days), I cannot open the webpage anymore (No response/timeout). I can still ping the ESP and the MQTT messages keep flowing. Only the webpage becomes unreachable until a hard reset.

I am now periodically opening a connection to the webserver from the ESP itself, to reset it if that does not work...

Pablo2048, 2018-06-13:
Sounds like memory fragmentation to me. If this is true then try to:
1. place all strings into PROGMEM
2. NEVER EVER use string concatenation with ordinary '+' - use string.reserve, followed with string.concat

klaasdc, 2018-06-13:
How can I check for memory fragmentation?
I never user String, only strcat/strcpy etc, but I do use ArduinoJson regularly inside the handlers.

Pablo2048, 2018-06-13:
Unfortunately IMHO You can't simply check memory fragmentation, because heap manager doesn't provide largest memory block size. Even if You use strcat/strcpy - do You have all const char [] in PROGMEM (are You using strcpy_P, strcat_P where possible?)?

Pablo2048, 2018-06-13:
Well, maybe is here way to check the heap - see https://github.com/esp8266/Arduino/blob/master/cores/esp8266/umm_malloc/umm_malloc.h there is unsigned short int maxFreeContiguousBlocks; inside UMM_HEAP_INFO_t ...

Pablo2048, 2018-06-13:
Probably You can turn on serial debug at least for TCP protocol - maybe this help...

klaasdc, 2018-06-14:
In the Arduino IDE, I only find debug messages for Core, http, etc but not specifically for lwIp. In any case, this does not seem to result in anything meaningful. I still have no clue where the problem could be.
I also noticed that when the webserver times out, in some case I can still ping the ESP, but sometimes not... The problem is reproducible but not consistent.

klaasdc, 2018-06-16:
Same problem when I load the ESP_AsyncFSBrowser example. Sometimes as soon as after 1 minute the ESP does not respond over WiFi anymore, sometimes it takes much longer.
I now fully believe the origin is somewhere "lower-level" than ESPAsyncWebServer, maybe related to https://github.com/esp8266/Arduino/issues/2330

closed 2019-10-05 by stale[bot]