me-no-dev/ESPAsyncWebServer #298: Timer0, sending spiffs and illegal instructions

judge2005, 2017-12-05:
Hi

I have an issue I am trying to solve. I have a timer0 interrupt handler that uses the SPI library. I also have a web server handler that sends data from a SPIFF. If the interrupt handler fires at the same time as the send is executing, I get an illegal instruction exception. This seems to be a known issue with accessing flash in interrupt handlers? Anyway, is there some way to be notified when a send has competed? Then maybe I can prevent my interrupt handler from being called for the duration of the send.

Of course, if there are better solutions, I am all ears.

I will also be looking to see if there might be better solutions to my issue than blocking the interrupt handler.

Thanks - Paul

atanisoft, 2017-12-08:
@judge2005 have you found any workarounds? I am hitting this same issue with timer0/timer1 ISRs firing at the same time as SPIFFs operations. I can't disable the ISRs since they are generating control signals on various pins.

judge2005, 2017-12-08:
@atanisoft The only solution I found was to make sure that all functions that were called from the interrupt handler were tagged with ICACHE_RAM_ATTR - including the interrupt handler itself. I had to copy some libraries and modify them to do that.

atanisoft, 2017-12-08:
@judge2005 I was hoping not to have to modify all of the dependencies :( Especially with limited IRAM. I guess SPIFFS/SD will be useless for this project. I will convert the html page (only one thankfully) to hex and embed in PROGMEM instead.

closed 2019-10-05 by stale[bot]