me-no-dev/ESPAsyncWebServer #299: AsyncTCP being pulled in for ESP8266 builds.

kylegordon, 2017-12-06:
Based on the reports at https://github.com/marvinroger/homie-esp8266/issues/444 https://travis-ci.org/kylegordon/homie-sonoff-touch/builds/312535275#L578 and https://travis-ci.org/euphi/Homie_BareMinimum/builds/312604604#L594 it would seem that AsyncTCP is being pulled in as a dependency for ESP8285 and ESP8266 builds.

I think, based on the commentary, that the ESPAsyncWebServer library (via PlatformIO) should only be requiring ESPAsyncTCP

atanisoft, 2017-12-07:
This is a dependency resolver bug in platformio. You can't define two different envs with different dependencies. I will dig up the bug report reference and add it shortly.

atanisoft, 2017-12-07:
platformio issue reference: https://github.com/platformio/platformio-core/issues/1153

It looks like they have fixed it via some adjustments to the platformio.ini file.

closed 2019-10-05 by stale[bot]