me-no-dev/ESPAsyncWebServer #300: Template variables' delimiter '%' conflicts with inline CSS

aharshac, 2017-12-08:
Hi @me-no-dev 
Thank you for an impressive library.

I'm trying to use a template processor on a file from SPIFFS. That file has inline CSS with '%' characters. The template processor considers any text including line breaks and whitespaces as templates. Entire chunks of CSS and other HTML markup are replaced by the default case of the template processor function.

For example, in the snippet below, 
``` html
<div style="background:#fff;height:50px;width:100%;display:flex;justify-content:center;align-items:center;z-index:90;position:absolute;top: 0;">
            <div style="width:100%;height:100%;display:flex;flex-direction:column;justify-content:center;align-items:left;">
```
`;display:flex;justify-content:center;align-items:center;z-index:90;position:absolute;top: 0;">
            <div style="width:100` is considered as a template variable.

Is it possible to not consider spaces while processing templates? Or rather, have an option to change the template variable delimiter?

atanisoft, 2017-12-08:
@aharshac you can change the template char from % to another single character here: https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/WebResponseImpl.h#L62

What might be good is switch from %VAR% to something like {VAR} instead with unique open/close characters around the variable name. This would likely be less problematic in HTML files (hopefully).

aharshac, 2017-12-08:
@atanisoft I've changed the template character to tilde ``` ` ```. Thank you.

While changing the template format or delimiter character would be excellent, simply avoiding text with spaces between two consecutive `%` is good enough for most cases.

closed 2019-10-05 by stale[bot]

chupocro, 2021-07-14:
This shouldn't be closed because not being able to use the percent character in the CSS templates makes template processing unusable. I've just spent 5 hours debugging the firmware because of that problem.

matrn, 2022-09-14:
Any updates to this issue? There is no readable ASCII character that cannot be normally contained in the html file so changing `TEMPLATE_PLACEHOLDER` is not a solution. It would be handy if more than one character in the separator will be supported and also start & end separator (for example for this `{{var}}`).

zekageri, 2022-09-14:
Open and end brackets would be the best approach

DeeEmm, 2022-11-17:
@zekageri & @matrn 

I agree. Open / close tagging along with double delimiters is standard practice for templating engines. 

Using a single character is ALWAYS going to create an issue for someone. Using a common valid character like % is a really poor choice.

It would not be too bad if that single character was user definable, at least the end user could change it on a case-by-case basis, but it's not possible to change it without hacking the core which means that it is not really a valid solution.

/DM

LondonJammo, 2023-04-07:
Still an issue, and ` % ` is still breaking the code. Was a fix/workaround found at all?

zekageri, 2023-04-07:
This lib is still not maintained anymore. If you want to change it, you need to tweak the library.

thueyts07, 2025-01-10:
Huge issue, making the feature unusable in any non-trivial case.
What I would do is changing the "Processor" definition so that it declares a list of placeholder to parse. The parser would ignore anything looking like %xyz% if xyz is not listed as a valid placeholder. Cheers.