me-no-dev/ESPAsyncWebServer #303: Need an easy way to send a binary buffer from memory

aguaviva, 2017-12-18:
Thanks!

Chinoforest, 2018-07-10:
Sorry, I'm needing the same. How can I send an array of uint8_t (or better yet, an array of uint16_t that's my desired type but if I can't get to send uint16_t directly I can get away with sending uint8_t)

I see there are PROGMEM functions like beginResponse_P but my data is in memory and I CAN NOT COPY it. I need it to send it without being copied.

fightingdreamer, 2018-11-17:
This should do:
```c++
server.on("/msgpack", HTTP_GET, [](AsyncWebServerRequest *request) {
  size_t size = 10;
  uint8_t data[size];
  AsyncResponseStream *response =
      request->beginResponseStream("application/msgpack", size);
  response->write(data, size);
  request->send(response);
};
```

closed 2019-10-05 by stale[bot]

hyansuper, 2022-06-27:
> This should do:
> 
> ```c++
> server.on("/msgpack", HTTP_GET, [](AsyncWebServerRequest *request) {
>   size_t size = 10;
>   uint8_t data[size];
>   AsyncResponseStream *response =
>       request->beginResponseStream("application/msgpack", size);
>   response->write(data, size);
>   request->send(response);
> };
> ```

I think there's still memory copy operation involved. 
see the [implementation of `response->write()`](https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/WebResponses.cpp#L692) is a [`cbuf->write()`](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/cbuf.cpp#L158) which copys memory.

So what's the best/efficient way to send binary data vie websocket?