me-no-dev/ESPAsyncWebServer #312: Hash dependency required in PlatformIO

kylegordon, 2018-01-15:
I think Hash needs to be declared as a dependency in http://platformio.org/lib/show/306/ESP%20Async%20WebServer as https://github.com/euphi/Homie_BareMinimum fails to build if you remove 'Hash' from the platformio.ini file.

```Compiling .pioenvs/esp01/lib124/ESP Async WebServer_ID306/AsyncEventSource.o
Compiling .pioenvs/esp01/lib124/ESP Async WebServer_ID306/AsyncWebSocket.o
Compiling .pioenvs/esp01/lib124/ESP Async WebServer_ID306/SPIFFSEditor.o
.piolibdeps/ESP Async WebServer_ID306/src/AsyncWebSocket.cpp:40:18: fatal error: Hash.h: No such file or directory
```

kimklai, 2018-11-03:
I found the problem. LDF failed to list correct library dep graph by default mode (chain mode). 
You should set 'lib_ldf_mode = deep+' to let LDF search all missing dep libraries. Or add "Hash" manually in lib_dep in case of LDF deep+ takes too long.

andig, 2019-05-28:
@me-no-dev it would still be nice if `Hash` could be listed as dependency inside `library.json`?

closed 2019-10-05 by stale[bot]