me-no-dev/ESPAsyncWebServer #313: Avoid to serv a specific file (config.json)

cotestatnt, 2018-01-16:
Hello.
I'm trying to avoid to serve  a specific file with both this methods, but it doesn't work.
With the browser I can always load the file.

```
// Avoid to load config file from browser
  server.on("/admin/config.json", HTTP_ANY, [](AsyncWebServerRequest *request) {    
    request->redirect("/admin");
  });
  
  server.on("/admin/config.json", HTTP_GET, [](AsyncWebServerRequest *request) {
    request->send(400);
  });
```

cotestatnt, 2018-01-16:
Arduino IDE 1.8.5 @ ESP8266 core 2.4.0 and latest release af ESPAsyncWebServer

penfold42, 2018-01-16:
Handlers are evaluated in the order they are attached to the server.

Did you set this before the spiffs handler ?

cotestatnt, 2018-01-16:
No after :'( 
Damn, it was so simple... thank you so much!

 :+1:

penfold42, 2018-01-16:
Happy to help.

Another approach is to serve a spiffs /www directory as / and keep your secret stuff out of there althogether

cotestatnt, 2018-01-17:
Finally, I used your second suggestion. Simple and secure at the same time.
Thak you, again.

closed 2019-10-05 by stale[bot]