me-no-dev/ESPAsyncWebServer #315: Problem with ESP8266 plugin version 2.4.0

UlliBien, 2018-01-21:
Hi,

I just installed the ESP8266 plugin version 2.4.0. Now the code does not compile any more :-(((

The first error is:
"ESPAsyncTCP.cpp: 113:36: error: macro "ip_route" requires 2 arguments, but only 1 given
   netif* interface = ip_route(&addr)"

There are a lot of other problems on lwIP. 
------------------------------

There are new lwIP options. There are four of them:
- v2 Prebuilt (MSS=536)
- v2 Prebuilt (MSS=1460, unstable)
- v1.4 Prebuilt
- v1.4 Open Source

The errors seem to be the same ones.

My knowledge on the subject is unfortunately very small. I can not fix the error by myself. I need your help.

Kind regards
ulli

dmsalo, 2018-01-22:
I had the same problem. I back to the version 2.4.0 rc1 so its works again.
Best regards.
DSalomon

UlliBien, 2018-03-07:
There a new version that compiles without errors. You should add a version history to your readme.md.

closed 2018-03-07 by UlliBien