me-no-dev/ESPAsyncWebServer #316: arduino-esp32 from 2018 doesn't work

emard, 2018-01-23:
HI

Something seems incompatible with arduino-esp32 from 2018
First commit in 2018 to arduino-esp32 makes SPIFFS no longer deliver
the page. ESPAsyncWebServer is still responsive to 192.168.4.1/heap
which is not in SPIFFS.

kristsm, 2018-01-24:
Hi! Just wanted to confirm that I have the same problems for delivering index.htm from SPIFFS. And it's not a problem to receive response from "heap". Are you planning to discover the problem? Thank you!

emard, 2018-01-24:
I guess a coordination of me-no-dev and ESP32 wifi team would be
the fastest way to solve this problem, I see from other posts that
me-no-dev must waits long time for esp32 wifi team to respond

my only solution for now is

 https://github.com/espressif/arduino-esp32
  git checkout 70d0d4648766cea047613062ed50a9ecfc9de31f

smacyas, 2018-01-29:
The same problem... https://github.com/espressif/arduino-esp32/issues/1022

vicdaner, 2018-01-30:
Same problem, sems that cannot charge javascript files

smacyas, 2018-01-31:
no comments from official persons....

vindolin, 2018-02-02:
I used fs.open from inside a sketch to write a file to SPIFFS and those files loaded without problems from a AsyncWebServerRequest, so I dug around which change broke this and found that if I used tools/sdk/lib/libspiffs.a from 70d0d464, files uploaded with the upload plugin work again.

smacyas, 2018-02-03:
@vindolin  Thanks! It works!

usmanshahid001, 2018-04-02:
@vindolin how to find specified plugin please help

closed 2019-10-05 by stale[bot]