me-no-dev/ESPAsyncWebServer #317: Not building on current versions.

vindolin, 2018-01-27:
After setting everything up from scratch following the installation instructions,
I get this error when trying to compile a blank project:
```
Compiling .pioenvs/esp32dev/lib37d/ESP Async WebServer/AsyncWebSocket.o
Archiving .pioenvs/esp32dev/libFrameworkArduino.a
In file included from .piolibdeps/ESP Async WebServer/src/AsyncWebSocket.h:30:0,
from .piolibdeps/ESP Async WebServer/src/AsyncWebSocket.cpp:22:
.piolibdeps/ESP Async WebServer/src/ESPAsyncWebServer.h:27:16: fatal error: FS.h: No such file or directory
```

closed 2018-01-27 by vindolin

theborax, 2018-02-16:
Did you solve this issue @vindolin ? I've also just set up a brand new project and run into this issue...

vindolin, 2018-02-19:
Yes I solved it, but I really have no clue what I did to make it work, I switched to Arduino, tried some things and came back to platform.io some days later and it just worked.

vindolin, 2018-02-19:
The only thing I did, was replacing tools/sdk/lib/libspiffs.a with the one from commit 70d0d4648766, but I don't think that's related.
See: https://github.com/me-no-dev/arduino-esp32fs-plugin/issues/6#issuecomment-362695983

theborax, 2018-02-19:
Think I figured it out. First I didn't include anything in my main file, but once I included FS.h things built. So building a "clean" main.cpp (or whatever one wants to call it) without any imports for i.e. FS.h wont work.

tomap, 2018-06-16:
Same for me. It did not work with the default main.cpp
I had to add 
```
#include <FS.h>
#include <WiFi.h>
```

NicHub, 2019-02-16:
Same for me too. You could update the [Installation/Using PlatformIO chapter](https://github.com/me-no-dev/ESPAsyncWebServer?utm_source=platformio&utm_medium=piohome#using-platformio).

    // Minimal main.cpp
    #include <Arduino.h>
    #include <FS.h>
    #include <WiFi.h>
    void setup() {}
    void loop() {}