me-no-dev/ESPAsyncWebServer #324: ESP32 corrupt heap when handling multiple simultaneous requests

GHUser1024, 2018-02-09:
Thank you for the great library, the speed and functionality are really amazing.

Attached file is basically a simplified webserver with 1 html page (which constantly refreshes itself) and several CSSs, all generated on the fly. 
This code works just fine on ESP8266, while ESP32 crashes with CORRUPT HEAP exception. 
Sometimes it happens after just a few refreshes, while sometimes the counter can go up to a few hundreds. But it's always reproducible.

[Test.ino.txt](https://github.com/me-no-dev/ESPAsyncWebServer/files/1709633/Test.ino.txt)

bbasil2012, 2018-02-09:
The same issue.

kherNI, 2018-02-11:
+1  Is this related?  Looks like a fix was published recently. https://esp32.com/viewtopic.php?f=2&t=4583

smacyas, 2018-03-13:
https://github.com/espressif/arduino-esp32/issues/1101

psykokwak-com, 2018-05-14:
+1 (And I use the latest espressif idf, arduino, tcpasync and asyncwebserver at the time I write this message)

ayushsharma82, 2018-05-19:
+1 happened with me too. It happened with ```send_P``` on simultaneous Refreshes and happens on random number of requests. ( Tested with latest IDE, Libraries )

iafilius, 2018-08-09:
+1, using latest arduino ide/esp32 library/compiler versions

me-no-dev, 2018-08-16:
pull latest AsyncTCP and the WebServer and try again

iafilius, 2018-08-17:
Hi, 

thanks for your update and work.
updated to the latest ESPwebserver and TCPAsync and tested:
1) the minimalistig ESPasync webserver and TCPAsync as posted in #324
2) my own software using ESPwebserver (in combination of Server Sent Events), TCPAsync.

1) was completely stable with initial tests (single client), and counter went over 1000, where is uses to crash within a few seconds mostly.
   however, with adding a second, 3th, 4th client, it crashed 
2) was much, much more stable, and but still crashes, which possible might be completely another issue.
I collected the crashes and crash decodings, not filtered on possible relevance or not

My impression, 
 some are relevant: tcp/multi_heap, updating SSE
 some not: ledCwrite
 
 Websockets has been updated/fixed, but SSE (which has similarities with websockets) not.
As stated i use Server Sent Events so possible an issue still exists in SSE.

Crash dumps and decodings (Core debug is set to "Error" in case it matters):

----
http refresh test code:

3493
3494
3495
CORRUPT HEAP: Bad head at 0x3ffbdf20. Expected 0xabba1234 got 0x00000000
E (289589) boot: Assert failed in multi_heap_free, /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/multi_heap_poisoning.c:205 (head != NULL)
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Core 1 register dump:
PC      : 0x400d98c5  PS      : 0x00060f34  A0      : 0x8008f3c4  A1      : 0x3ffbc3f0  
A2      : 0x3ffc3a58  A3      : 0x000000cd  A4      : 0x3ffc3b1c  A5      : 0x3ffc3a48  
A6      : 0x00060d23  A7      : 0x00000000  A8      : 0x800d98c5  A9      : 0x3ffbc3a0  
A10     : 0x400ea08c  A11     : 0x3ffc4e18  A12     : 0x3f4024d0  A13     : 0x00000008  
A14     : 0x00000002  A15     : 0x00000001  SAR     : 0x00000004  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffc  

Backtrace: 0x400d98c5:0x3ffbc3f0 0x4008f3c1:0x3ffbc420 0x40084782:0x3ffbc440 0x40086949:0x3ffbc460 0x4000bec7:0x3ffbc480 0x400d74ba:0x3ffbc4a0 0x400d5225:0x3ffbc4c0 0x400d5256:0x3ffbc4e0 0x400d5262:0x3ffbc500 0x400d4a16:0x3ffbc520 0x400d5861:0x3ffbc540 0x400d2827:0x3ffbc560 0x400d2835:0x3ffbc580 0x400d62ee:0x3ffbc5a0 0x400d63f7:0x3ffbc5d0 0x400d64a5:0x3ffbc620 0x400d6731:0x3ffbc660

Core 0 register dump:
PC      : 0x4008cf7e  PS      : 0x00060e34  A0      : 0x8008ecfe  A1      : 0x3ffb6fc0  
A2      : 0x3ffae718  A3      : 0x0000cdcd  A4      : 0xb33fffff  A5      : 0x00000001  
A6      : 0x00060e20  A7      : 0x0000abab  A8      : 0x0000cdcd  A9      : 0x000000ff  
A10     : 0x3ffc7cbc  A11     : 0x00000001  A12     : 0x3ffc9a08  A13     : 0x00000000  
A14     : 0x00000070  A15     : 0x3ffc208c  SAR     : 0x00000019  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x4008cf7e:0x3ffb6fc0 0x4008ecfb:0x3ffb6ff0 0x4008f373:0x3ffb7010 0x40084688:0x3ffb7030 0x4008473e:0x3ffb7050 0x40083ded:0x3ffb70a0 0x401227a7:0x3ffb70c0 0x4012c988:0x3ffb7100 0x40088ab1:0x3ffb7180

--
CPU 1:

PC: 0x400d98c5: __assert_func at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bootloader_support/src/bootloader_init.c line 523
EXCVADDR: 0x00000000

Decoding stack results
0x400d98c5: __assert_func at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bootloader_support/src/bootloader_init.c line 523
0x4008f3c1: multi_heap_free at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/multi_heap_poisoning.c line 205
0x40084782: heap_caps_free at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/heap_caps.c line 262
0x40086949: _free_r at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/newlib/syscalls.c line 42
0x400d74ba: String::~String() at C:\Users\m99i941\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\cores\esp32\WString.cpp line 129
0x400d5225: AsyncWebServerResponse::~AsyncWebServerResponse() at C:\Users\m99i941\Arduino\libraries\ESPAsyncWebServer\src\WebResponses.cpp line 103
0x400d5256: AsyncBasicResponse::~AsyncBasicResponse() at C:\Users\m99i941\Arduino\libraries\ESPAsyncWebServer\src/WebResponseImpl.h line 32
0x400d5262: AsyncBasicResponse::~AsyncBasicResponse() at C:\Users\m99i941\Arduino\libraries\ESPAsyncWebServer\src/WebResponseImpl.h line 32
0x400d4a16: AsyncWebServerRequest::~AsyncWebServerRequest() at C:\Users\m99i941\Arduino\libraries\ESPAsyncWebServer\src\WebRequest.cpp line 87
0x400d5861: AsyncWebServer::_handleDisconnect(AsyncWebServerRequest*) at C:\Users\m99i941\Arduino\libraries\ESPAsyncWebServer\src\WebServer.cpp line 97
0x400d2827: AsyncWebServerRequest::_onDisconnect() at C:\Users\m99i941\Arduino\libraries\ESPAsyncWebServer\src\WebRequest.cpp line 226
0x400d2835: std::_Function_handler   >::_M_invoke(const std::_Any_data &,  ,  ) at C:\Users\m99i941\Arduino\libraries\ESPAsyncWebServer\src\WebRequest.cpp line 73
0x400d62ee: std::function ::operator()(void*, AsyncClient*) const at c:\users\m99i941\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0/functional line 2271
[... 261 lines cut ...]
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d9aa4  PS      : 0x00060130  A0      : 0x800d9b08  A1      : 0x3ffb1e50  
A2      : 0x3ffaf5a8  A3      : 0x3ffaff08  A4      : 0x48000000  A5      : 0x3ffaff00  
[... cut ...]
0x400d9b63: AsyncEventSource::send(char const*, char const*, unsigned int, unsigned int) at C:\Users\m99i941\Arduino\libraries\ESPAsyncWebServer\src\AsyncEventSource.cpp line 298
0x400d4a38: update_html5_sse_values(bool) at C:\Users\m99i941\Arduino\DS18B20_PWMFAN_aVR_ESP32_WiFI_BLE_20180812_ESP/async_http_server.ino line 1036
0x400d8299: loop() at C:\Users\m99i941\Arduino\DS18B20_PWMFAN_aVR_ESP32_WiFI_BLE_20180812_ESP/DS18B20_PWMFAN_aVR_ESP32_WiFI_BLE_20180812_ESP.ino line 914
0x4013b79a: loopTask(void*) at C:\Users\m99i941\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\cores\esp32\main.cpp line 17

Regards,

me-no-dev, 2018-08-17:
all corrupt heap exceptions are caused by the same thing that I have been hunting... function accesses a freed client. multitasking issue.

me-no-dev, 2018-08-17:
just pushed another change to async tcp. now i generally get the errors elsewhere... like while closing a file being served and such... please whoever can test and see whet differences does it get. try normal operation too, fast reload before a page is loaded is quite the stress...

iafilius, 2018-08-18:
Hi, fetched the last updated version (5453ec2e), and performed some tests.

1) refresh demo/test
2) my own progam. (using Server Sent Events)

summary/findings/conclusions:

All crashdumps below have a short description on which test and remarks.

Exception decoder seems not to look at both crash dumps of Core 1 and 0 (if provided). Need to paste Core 1 dump and  afterwards Core 0 dump. This is what I did in cases crashdumps of both core's where given.  And I think it may provide additional relevant information.

-1) http refresh test
	A) A single http refresh session , seems to be rock stable, no crashes (as previous commit)
	B) I was able to have 10 concurrent refresh sessions (tabs in firefox) from a single device (read wifi client), just a single crash but seems rather stable. (didn't test above 10)
	C) Having 1 http refresh session from one wifi client device, and having one refresh session from another wifi client device, a quite reproducable crash happens mostly instant, after starting another refresh session on an existing wifi client device (second tab in firefox).
		-> Looking at the crashdumps (multiple of this exact situation) it looks like a heap free/alloc race condition or locking issue (wifi is allocating memory (Core 0), while asynctcp is free-ing some (core 1)).  (I think this would imply a shared heap is used between Core 1 and 0)
-2) my own program:
	A) seems to have an issue with ledC write in a timer interrupt, didn't notice before yesterday, but it might not related to tcp/close/heap corruption. I guess it was masked by more frequent crashes (started arduino board manager for ESP32 recently as well ).
	B) the occasional restarting webclient (which resulted frequently in a crash) seems to be solved with latest updates, need to push much harder (2C).
	C) has still some issues when wildly stopping/starting the webapp/client , and it is harder to trigger then with previous.
		-> as with 1C, looks like free/alloc race condition between CPU0/1.

	
	
----
Crash 1:
My own software testing PWM (ledC) on/off updates:

/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/queue.c:1445 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x4008d32b on core 1

Backtrace: 0x40090570:0x3ffc0b30 0x40090773:0x3ffc0b50 0x4008d32b:0x3ffc0b70 0x400e367f:0x3ffc0bb0 0x40081093:0x3ffc0bd0 0x400820c1:0x3ffc0bf0 0x400828d5:0x3ffc0c10 0x4000bfed:0x00000000

--

Decoding stack results
0x40090570: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 140
0x40090773: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 149
0x4008d32b: xQueueGenericReceive at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/queue.c line 1445
0x400e367f: ledcWrite at C:\Users\m99i941\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\cores\esp32\esp32-hal-ledc.c line 164
0x40081093: PWM_control_Timer_callback() at C:\Users\m99i941\Arduino\DS18B20_PWMFAN_aVR_ESP32_WiFI_BLE_20180812_ESP/DS18B20_PWMFAN_aVR_ESP32_WiFI_BLE_20180812_ESP.ino line 543
0x400820c1: __timerISR at C:\Users\m99i941\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\cores\esp32\esp32-hal-timer.c line 88

----
Crash 2:
testing own software: repeated starting (single) webapp (using SSE) and killing it:

g_Enforce_HTTP5_SSE_update !!!! updating all paramters to all clients
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d9ae4  PS      : 0x00060130  A0      : 0x800d9b48  A1      : 0x3ffb1e50  
A2      : 0x3ffb545c  A3      : 0x3ffbfff8  A4      : 0x20abba12  A5      : 0x3ffbf8f4  
A6      : 0x0000000a  A7      : 0x323a3430  A8      : 0x800d9af0  A9      : 0x3ffb1e30  
A10     : 0x00000001  A11     : 0x3ffbf940  A12     : 0x89abba12  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000009  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x400d9ae4:0x3ffb1e50 0x400d9b45:0x3ffb1e70 0x400d9b63:0x3ffb1e90 0x400d9ba3:0x3ffb1eb0 0x400d4a38:0x3ffb1ee0 0x400d8299:0x3ffb1f80 0x4013b7da:0x3ffb1fa0

Rebooting...

--
PC: 0x400d9ae4: AsyncEventSourceClient::_runQueue() at C:\Users\m99i941\Arduino\libraries\ESPAsyncWebServer\src\AsyncEventSource.cpp line 239
EXCVADDR: 0x00000009

Decoding stack results
0x400d9ae4: AsyncEventSourceClient::_runQueue() at C:\Users\m99i941\Arduino\libraries\ESPAsyncWebServer\src\AsyncEventSource.cpp line 239
0x400d9b45: AsyncEventSourceClient::_queueMessage(AsyncEventSourceMessage*) at C:\Users\m99i941\Arduino\libraries\ESPAsyncWebServer\src\AsyncEventSource.cpp line 189
0x400d9b63: AsyncEventSourceClient::write(char const*, unsigned int) at C:\Users\m99i941\Arduino\libraries\ESPAsyncWebServer\src\AsyncEventSource.cpp line 224
0x400d9ba3: AsyncEventSource::send(char const*, char const*, unsigned int, unsigned int) at C:\Users\m99i941\Arduino\libraries\ESPAsyncWebServer\src\AsyncEventSource.cpp line 298
[... 554 lines cut ...]
0x400d49b1: AsyncWebServerRequest::~AsyncWebServerRequest() at C:\Users\m99i941\Arduino\libraries\ESPAsyncWebServer\src/StringArray.h line 166
0x400d5861: AsyncWebServer::_handleDisconnect(AsyncWebServerRequest*) at C:\Users\m99i941\Arduino\libraries\ESPAsyncWebServer\src\WebServer.cpp line 97
[... cut ...]
0x4008f373: multi_heap_malloc at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/multi_heap_poisoning.c line 181
0x40084688: heap_caps_malloc at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/heap_caps.c line 103
0x4008473e: heap_caps_malloc_prefer at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/heap_caps.c line 181
0x40083ded: wifi_malloc at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/wifi_os_adapter.c line 61
----

Regards, Arjan

me-no-dev, 2018-08-18:
yeah, i am getting similar results, but have really hard time tracing them. I need to find the cause before I can fix it and that is proving impossible so far

hostmit, 2018-09-19:
any news on this issue? having same problem

HenrikSte, 2018-09-30:
yea, same with me. What a pity. 
Had to switch back to the sync server as I was not able to get this running stable even with the example code...

hostmit, 2018-10-01:
@me-no-dev this issue warning SHOULD be on main page. Ppl like me wasting time, implementing logic on ESPASYNC, then finding out it crashes on ESP32....

me21, 2018-10-01:
There is a single-core variant of ESP32. I wonder if it crashes there. Anyone?

smacyas, 2018-10-01:
Yes, it still heap corrupts and crashes

dmars1972, 2018-10-04:
I'm getting the same issue with AsyncUDP, if it helps.

davidgevorgian, 2018-10-09:
Same problem, Update.write(data, len) crash.....

davidgevorgian, 2018-11-13:
Multi clients = CRASH

CORRUPT HEAP: Bad head at 0x3ffcb550. Expected 0xabba1234 got 0x3ffcb688
assertion "head != NULL" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/multi_heap_poisoning.c", line 214, function: multi_heap_free
abort() was called at PC 0x400e4d6f on core 1

Decoding stack results
0x4008f9c4: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 155
0x4008fb99: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 170
0x400e4d6f: __assert_func at ../../../.././newlib/libc/stdlib/assert.c line 63
0x4008f691: multi_heap_free at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/multi_heap_poisoning.c line 214
0x40084872: heap_caps_free at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/heap_caps.c line 269
0x40086a31: _free_r at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/newlib/syscalls.c line 42
0x40105b6f: mem_free at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/mem.c line 151
0x40105bcf: do_memp_free_pool at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/memp.c line 432
0x40105c17: memp_free at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/memp.c line 489
0x40108336: tcp_close_shutdown at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/tcp.c line 311
0x4010839e: tcp_close at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/tcp.c line 409
0x400d53fd: _tcp_close_api(tcpip_api_call_data*) at C:\Users\David\Documents\Arduino\libraries\AsyncTCP\src\AsyncTCP.cpp line 346
0x401041c0: tcpip_thread at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/api/tcpip.c line 124
0x4008ba0d: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 141

CelliesProjects, 2018-12-18:
Still no news on the debugging front?

I could post some debug output, but that boils down to already posted errors.
Specifically this one pops up:
```
multi_heap_free at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/multi_heap_poisoning.c line 214
```
Will keep an aye on this thread.

hostmit, 2019-02-21:
Any solution?

psykokwak-com, 2019-02-21:
Now I use mangoose ( https://github.com/cesanta/mongoose ) that works very well :)

JacoFourie, 2019-02-21:
@psykokwak-com do you have a sample using that lib in Arduino IDE ?

hostmit, 2019-02-21:
:( I was happy... for like 30 sec. they dont have it ported for Arduino... only IDF :((((

psykokwak-com, 2019-02-21:
Arduino layer is over IDF. You can use it directly without problem with the arduino layer. This is what I do.

psykokwak-com, 2019-02-21:
Just include the .h file then put initialization stuf in setup() and the mg_mgr_poll(&mgr, 1); function in loop()

JacoFourie, 2019-02-21:
@psykokwak-com can you post a little sample of how you use it ?

jeremypoulter, 2019-06-22:
FYI I have started a project to create a wrapper around Mongoose for Arduino* you can see my progress [here](https://github.com/jeremypoulter/ArduinoMongoose). My goal is to create an API similar to ESPAsyncWebServer, but is not a drop in replacement as I want the wrapper to be fairly light weight.

Grateful for any help.

 * I use the term Arduino for the APIs not the IDE, I use Platform IO primarily so have not focused on it working with Arduino IDE

me-no-dev, 2019-06-22:
currently looking into this issue. Some good progress so far

me-no-dev, 2019-06-22:
alright... AsyncTCP has been updated. Please give it a shot. FYI, the issue has never been in the WebServer :)

jeremypoulter, 2019-06-26:
Thanks, did give it a try, no crash, but do now get a watchdog timeout

```
E (188784) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (188784) task_wdt:  - async_tcp (CPU 0/1)
E (188784) task_wdt: Tasks currently running:
E (188784) task_wdt: CPU 0: IDLE0
E (188784) task_wdt: CPU 1: loopTask
E (188784) task_wdt: Aborting.
abort() was called at PC 0x400f9238 on core 0

Backtrace: 0x4008cbcc:0x3ffbe160 0x4008cdfd:0x3ffbe180 0x400f9238:0x3ffbe1a0 0x40081b59:0x3ffbe1c0 0x4017a3a7:0x3ffbce80 0x400f7746:0x3ffbcea0 0x4008aa69:0x3ffbcec0 0x40088b69:0x3ffbcee0
```

me-no-dev, 2019-06-26:
please decode that backtrace. It is an issue in your code :) WDT was added to catch those cases.

zekageri, 2019-07-01:
This issue is gone in STA mode for me, but in STA+AP mode i get a reboot with the following messages:

`dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
dhcps: send_nak>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
dhcps: send_offer>>udp_sendto result 0
assertion "new_rcv_ann_wnd <= 0xffff" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/tcp.c", line 779, function: tcp_update_rcv_ann_wnd
abort() was called at PC 0x4010123b on core 0

Backtrace: 0x4008d860:0x3ffb3d70 0x4008da91:0x3ffb3d90 0x4010123b:0x3ffb3db0 0x40127d72:0x3ffb3de0 0x40127e01:0x3ffb3e00 0x40154c05:0x3ffb3e20 0x401247a0:0x3ffb3e40 0x4008977d:0x3ffb3e70`

Decoded:

`0x4008d860: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 155
0x4008da91: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 170
0x4010123b: __assert_func at ../../../.././newlib/libc/stdlib/assert.c line 63
0x40127d72: tcp_update_rcv_ann_wnd at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/tcp.c line 779
0x40127e01: tcp_recved at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/tcp.c line 820
0x401247a0: tcpip_thread at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/api/tcpip.c line 124
0x4008977d: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143
`

zekageri, 2019-07-03:
Sometimes i get the muti heap problem in AP mode, but never in STA mode anymore. If i use soft ap while my esp32 connected to the network via ethernet, the AP and the Ethernet connection is fast as it should be, but if i try to load a page three or more times in AP mode it crashes with this error:

```
dhcps: send_offer>>udp_sendto result 0
assertion "next > (intptr_t)block" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/multi_heap.c", line 124, function: get_next_block
abort() was called at PC 0x401011ff on core 0

Backtrace: 0x4008d860:0x3ffb3cd0 0x4008da91:0x3ffb3cf0 0x401011ff:0x3ffb3d10 0x4008d05b:0x3ffb3d40 0x4008d4de:0x3ffb3d60 0x40085a56:0x3ffb3d80 0x4008723d:0x3ffb3da0 0x4000bec7:0x3ffb3dc0 0x40125d6f:0x3ffb3de0 0x40125dcf:0x3ffb3e00 0x40125e17:0x3ffb3e20 0x40124776:0x3ffb3e40 0x4008977d:0x3ffb3e70
```

Decoded:

```
Decoding stack results
0x4008d860: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 155
0x4008da91: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/panic.c line 170
0x401011ff: __assert_func at ../../../.././newlib/libc/stdlib/assert.c line 63
0x4008d05b: multi_heap_free_impl at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/multi_heap.c line 124
0x4008d4de: multi_heap_free at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/multi_heap_poisoning.c line 221
0x40085a56: heap_caps_free at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/heap_caps.c line 268
0x4008723d: _free_r at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/newlib/syscalls.c line 42
0x40125d6f: mem_free at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/mem.c line 151
0x40125dcf: do_memp_free_pool at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/memp.c line 432
0x40125e17: memp_free at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/memp.c line 489
0x40124776: tcpip_thread at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/api/tcpip.c line 139
0x4008977d: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143
```

closed 2019-10-05 by stale[bot]

LukaBronzovic, 2019-11-25:
Has the issue with the corrupt heap been fixed by now? I am getting the following with one client and quite slow client updates (every 3 seconds via WebSocket). The error happenswhen i try to free() the memory allocated with malloc().
___________________________________________________________________________________________________________
CORRUPT HEAP: Bad tail at 0x3ffd3fb9. Expected 0xbaad5678 got 0xbaad5600
assertion "head != NULL" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c", line 214, function: multi_heap_free
abort() was called at PC 0x40118a23 on core 1

Backtrace: 0x4008c434:0x3ffd2ef0 0x4008c665:0x3ffd2f10 0x40118a23:0x3ffd2f30 0x4008c0a9:0x3ffd2f60 0x40084b12:0x3ffd2f80 0x40084f19:0x3ffd2fa0 0x4000bec7:0x3ffd2fc0 0x400d2001:0x3ffd2fe0 0x40178843:0x3ffd3020 0x40178cce:0x3ffd3040 0x400d6c71:0x3ffd3090 0x400d6c8d:0x3ffd30d0 0x40168a05:0x3ffd30f0 0x40168a81:0x3ffd3130 0x40169126:0x3ffd3150 0x40088b7d:0x3ffd3180

Rebooting...

___________________________________________________________________________________________________________

Library versions I am using are:

AsynchTCP 1.1.1
ESPAsyncWebServer 1.2.3

thbl, 2020-08-06:
+1 @me-no-dev

zekageri, 2020-11-11:
> Has the issue with the corrupt heap been fixed by now? I am getting the following with one client and quite slow client updates (every 3 seconds via WebSocket). The error happenswhen i try to free() [...]

Exactly the same.

psykokwak-com, 2020-12-21:
The best solution is to use another library like https://github.com/cesanta/mongoose

jeremypoulter, 2020-12-21:
For what it is worth I have created an Arduino wrapper for mongoose: https://github.com/jeremypoulter/ArduinoMongoose it has a similar API to ESPAsyncWebServer so is an easy port

themindfactory, 2020-12-21:
not the best place to ask, but can this serve up multiple files at the same
time? Just like async lib does?

Thanks!

psykokwak-com, 2020-12-21:
Yes. (At least on ESP32). It use the BSD file API.

BarmerO, 2021-02-23:
I got an error and it is similar to the one described here. How can you solve the problem?

CORRUPT HEAP: Bad head at 0x3ffd631c. Expected 0xabba1234 got 0x00000000
assertion "head != NULL" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c", line 214, function: multi_heap_free
abort() was called at PC 0x40116f73 on core 1

Backtrace: 0x4008c8c0:0x3ffd5e70 0x4008caf1:0x3ffd5e90 0x40116f73:0x3ffd5eb0 0x4008c535:0x3ffd5ee0 0x40084f9e:0x3ffd5f00 0x400853a5:0x3ffd5f20 0x4000bec7:0x3ffd5f40 
0x4014f019:0x3ffd5f60 0x400d62ad:0x3ffd5f80 0x400d820b:0x3ffd5fa0 0x400d8a89:0x3ffd5fc0 0x400daedd:0x3ffd5fe0 0x400d6423:0x3ffd6000 0x400d642f:0x3ffd6020 0x4016b302:0x3ffd6040 0x4016b436:0x3ffd6070 0x4016b44c:0x3ffd6090 0x4016b70e:0x3ffd60b0 0x40089009:0x3ffd60e0

zekageri, 2021-02-23:
You can do nothing. Esp websocket crashing everywhere. You can not hide from it.

BarmerO, 2021-02-24:
Thanks for the answer. Very sorry. And there are examples of using Mongose with the file system. I can't find it on the internet.