me-no-dev/ESPAsyncWebServer #326: ESP32: md5.h: No such file or directory

probonopd, 2018-02-11:
Getting

```
ESP32: ESPAsyncWebServer/src/WebAuthentication.cpp:23:17: fatal error: md5.h: No such file or directory
```

when trying to compile on ESP32.

Reference:
https://github.com/probonopd/WirelessPrinting/issues/29

argo69, 2018-03-02:
Hello, 
I have the same error message "md5.h: no such file or directory" when trying to compile on ESP32.
"blink" sketch compile and works ok on the board ...
An idea ?

argo69, 2018-03-02:
of course i've checked the file is present on the correct directory...

closed 2019-10-05 by stale[bot]

probonopd, 2019-10-05:
@stale go away

me-no-dev, 2019-10-05:
@probonopd this library get's tested against Arduino and PIO on Linux, Windows and Mac. There are no dependency issues. This is not an issue in this library.

probonopd, 2019-10-05:
OK thanks @me-no-dev, this is actually a usable information (other than the Stale bot just closing stuff).

me-no-dev, 2019-10-05:
stale bot closed it because nobody responded for 2 weeks after it labeled the message ;) see the post it left.