me-no-dev/ESPAsyncWebServer #327: Update example for ESP32

probonopd, 2018-02-11:
Please update `` with something along the lines of

```
#if defined(ESP8266)
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <ESPAsyncTCP.h>
#endif

#if defined(ESP32)
#include <WiFi.h>
#include <ESPmDNS.h>
#include <AsyncTCP.h>
#endif
```

However it does not work for me when I do the above:

```
ESP_AsyncFSBrowser.ino:15:18: fatal error: Hash.h: No such file or directory
compilation terminated.
Multiple libraries were found for "WiFi.h"
 Used: /home/me/Arduino/hardware/espressif/esp32/libraries/WiFi
 Not used: /tmp/.mount_qjlq9N/usr/bin/libraries/WiFi
exit status 1
Error compiling for board ESP32 Dev Module.
```

rokrodic, 2018-02-21:
Hi. I use latest ESP32 core, latest Arduino IDE.
I receive the same errors as above. I cannot find "Hash.h" file?!? I defined the same "if"s above. The other errors I get are:
C:\Users\AKOYA\AppData\Local\Temp\arduino_modified_sketch_902778\ESP_AsyncFSBrowser.ino: In function 'void setup()':
ESP_AsyncFSBrowser:106: error: 'class WiFiClass' has no member named 'hostname'
   WiFi.hostname(hostName);
        ^
ESP_AsyncFSBrowser:137: error: 'SPIFFS' was not declared in this scope
   SPIFFS.begin();
   ^
ESP_AsyncFSBrowser:147: error: no matching function for call to 'SPIFFSEditor::SPIFFSEditor(const char*&, const char*&)'
   server.addHandler(new SPIFFSEditor(http_username,http_password));
                                                                 ^
In file included from C:\Users\AKOYA\AppData\Local\Temp\arduino_modified_sketch_902778\ESP_AsyncFSBrowser.ino:17:0:
C:\Users\AKOYA\Documents\Arduino\Sketchbook\hardware\espressif\esp32\libraries\ESPAsyncWebServer\src/SPIFFSEditor.h:14:5: note: candidate: SPIFFSEditor::SPIFFSEditor(const fs::FS&, const String&, const String&)
     SPIFFSEditor(const fs::FS& fs, const String& username=String(), const String& password=String());
     ^
C:\Users\AKOYA\Documents\Arduino\Sketchbook\hardware\espressif\esp32\libraries\ESPAsyncWebServer\src/SPIFFSEditor.h:14:5: note:   no known conversion for argument 1 from 'const char*' to 'const fs::FS&'
C:\Users\AKOYA\Documents\Arduino\Sketchbook\hardware\espressif\esp32\libraries\ESPAsyncWebServer\src/SPIFFSEditor.h:5:7: note: candidate: SPIFFSEditor::SPIFFSEditor(const SPIFFSEditor&)
 class SPIFFSEditor: public AsyncWebHandler {
       ^
C:\Users\AKOYA\Documents\Arduino\Sketchbook\hardware\espressif\esp32\libraries\ESPAsyncWebServer\src/SPIFFSEditor.h:5:7: note:   candidate expects 1 argument, 2 provided
C:\Users\AKOYA\Documents\Arduino\Sketchbook\hardware\espressif\esp32\libraries\ESPAsyncWebServer\src/SPIFFSEditor.h:5:7: note: candidate: SPIFFSEditor::SPIFFSEditor(SPIFFSEditor&&)
C:\Users\AKOYA\Documents\Arduino\Sketchbook\hardware\espressif\esp32\libraries\ESPAsyncWebServer\src/SPIFFSEditor.h:5:7: note:   candidate expects 1 argument, 2 provided
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\AKOYA\Documents\Arduino\Sketchbook\hardware\espressif\esp32\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi

Any idea?

jesse-schein, 2018-03-20:
I think this may be fixed, at least I have had luck including some other library on PlatformIO to build with async webserver no problems.Running very well on the ESP32.

dsteinman, 2018-04-19:
It's not fixed yet for Arduino IDE.  I was still getting `Hash.h file not found` in Arduino IDE.

Using PlatformIO I was able to compile the `ESP_AsyncFSBrowser.ino` example to work with the following changes to the includes:

```
#include <Arduino.h>
//#include <Hash.h>
//#include <ESP8266WiFi.h>
#include <WiFi.h>
//#include <ESP8266mDNS.h>
#include <ESPmDNS.h>
#include <ArduinoOTA.h>
#include <FS.h>

//#include <ESPAsyncTCP.h>
#include <AsyncTCP.h>

#include <SPIFFS.h>
//#include <SPIFFSEditor.h>
//#include <AsyncWebSocket.h>
#include <ESPAsyncWebServer.h>
```

And I changed `WiFi.hostname` to `WiFi.setHostname` and moved it after `.begin()` to avoid [this issue](https://github.com/espressif/arduino-esp32/issues/806):

```
WiFi.mode(WIFI_AP_STA);
  WiFi.softAP(hostName);
  WiFi.begin(ssid, password);
  WiFi.setHostname(hostName);
```

And I added a line to echo the IP address:
```
 Serial.print ( "IP address: " );
 Serial.println ( WiFi.localIP() );
```

And changed the ssid and password for my router:

```
const char* ssid = "****";
const char* password = "********";
```

In PlatformIO's library panel I installed the 'ESP Async WebServer' library and edited `platformio.ini`:

```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps =
  SPIFFS
  AsyncTCP
  ESP Async WebServer
```

I have to use the [arduino-esp32fs-plugin](https://github.com/me-no-dev/arduino-esp32fs-plugin) `ESP Sketch Data Upload` tool in Arduino to upload my /data directory.  Then uploaded the sketch using PlatformIO's Upload button.

So the only drawback is I need to use both PlatformIO and Arduino until the esp32fs-plugin tool can be integrated into PlatformIO.

Full gist:
https://gist.github.com/dsteinman/f792f0af25ce6d7d1db4b62d29dd4d9e

closed 2019-10-05 by stale[bot]

probonopd, 2019-10-05:
@stale go away

me-no-dev, 2019-10-05:
this is fixed.