me-no-dev/ESPAsyncWebServer #330: ESP32 SPIFFS Editor just spins

terryjmyers, 2018-03-03:
It seems to almost work, but its just spinning.
`#include <WiFi.h>
#include <AsyncTCP.h>
#include <SPIFFS.h>
#include <SPIFFSEditor.h>

AsyncWebServer HTTPserver(80);

const char* http_username = "admin";
const char* http_password = "admin";

void setup()
{
	Serial.begin(115200);
	
	SPIFFS.begin();

	WiFi.begin();
	WiFi.waitForConnectResult();
	Serial.println(WiFi.localIP().toString());

	HTTPserver.addHandler(new SPIFFSEditor(SPIFFS, http_username, http_password));
	HTTPserver.begin();	
}
void loop()
{
}

`
I've made sure I've gotten the latest ESP Arduino Core, and the latest Libraries for each.  Is is broken?

Picture of the webpage:
[https://imgur.com/a/Jssgz](url)

Pablo2048, 2018-03-03:
What is the result from SPIFFS.begin()?

terryjmyers, 2018-03-03:
Ok....well I just erased all of the files in SPIFFS by using "ESP32 SPIFFS Upload Tool" with an empty data folder, and now its working as expected.

Just for shits and giggles I flashed it again with all of the files in the ESP_AsyncFSBrowser and this is what is confusing it.  In other words, if you have the files in "data" provided by the example in ESP_AsyncFSBrowser it doesn't work.

closed 2018-03-03 by terryjmyers