me-no-dev/ESPAsyncWebServer #331: On chrome, streaming temperature over websocket fails with error 1006 after a while

aguaviva, 2018-03-03:
This is from the Sousvide project (https://github.com/aguaviva/SousVide)

I am including a wireshark capture of when things go wrong, here is also an screenshot:

[image]

You can repro the issue without any hardware, just clone the repo and in main.cpp substitute these lines:

    static double Input;
    temperatureGetReadingSecurity(10, &Input);

with

    static double Input=10;

thats' it.

aguaviva, 2018-03-04:
the latest development version solves this issue, I'd be interesting to understand what was causing that, it might be related to issue #332 (https://github.com/me-no-dev/ESPAsyncWebServer/issues/332)

aguaviva, 2018-03-06:
The issue is still happening, it just takes longer to occur. 

Is there anything else I could do to help debug this issue?  I'd appreciate some guidance.

aguaviva, 2018-04-09:
@espressif guys, I'd appreciate some guidance on how to fix this, I am sure I am not the only one trying to use sockets to stream data...

aguaviva, 2019-10-04:
shame

closed 2019-10-04 by aguaviva

RamonHartmann, 2023-07-21:
same problem