me-no-dev/ESPAsyncWebServer #337: Send a file using ESP8266HTTPClient (http.sendRequest) to another ESP8266 running ESPAsyncWebServer (POST method)

hagelric, 2018-03-16:
I'm trying to send a file (100kb) using two ESP8266... One is running HTTPClient to send the file, another is running ESPAsyncWebserver. 
When I try upload using chrome, all works fine. But not works when I send using code below:

How can I fix? _or if there is a suggestion to use another code to do it..._

**(A) ESP8266 (running ESP8266HTTPClient)**
```      
      nameFile = "/test.htm";
      fs::File fsUploadFile2 = SPIFFS.open(nameFile, "r")                      
      String MPUdataAddress = "http://192.168.25.51/auth/MPUdata";
      http.begin(MPUdataAddress);
      http.addHeader("Content-Type", "multipart/form-data");
      int httpCode = http.sendRequest("FILE", &fsUploadFile2, fsUploadFile2.size());
      if (httpCode >= 200 && httpCode < 300)  {
         String payload2 = http.getString();
         Serial.println("SUCCESS! DATA RECEIVED: ");
         Serial.println(payload2);       
      }
      http.end();
      fsUploadFile2.close();
```

(A) HTTP Header DEBUG

```
HTTP-Client][begin] url: http://192.168.25.51/auth/MPUdata
[HTTP-Client][begin] host: 192.168.25.51 port: 80 url: /auth/MPUdata
[HTTP-Client] connected to 192.168.25.51:80
[HTTP-Client] sending request header
-----
FILE /auth/MPUdata HTTP/1.1
Host: 192.168.25.51
User-Agent: ESP8266HTTPClient
Connection: close
Accept-Encoding: identity;q=1,chunked;q=0.1,*;q=0
Content-Type: multipart/form-data
Content-Length: 2088

-----
[HTTP-Client][sendRequest] Stream payload written: 2088
[HTTP-Client][handleHeaderResponse] RX: 'HTTP/1.1 200 OK'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Length: 3'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Type: text/plain'
[HTTP-Client][handleHeaderResponse] RX: 'Connection: close'
[HTTP-Client][handleHeaderResponse] RX: 'Connection: Keep-Alive'
[HTTP-Client][handleHeaderResponse] RX: 'Accept-Ranges: none'
[HTTP-Client][handleHeaderResponse] RX: ''
[HTTP-Client][handleHeaderResponse] code: 200
[HTTP-Client][handleHeaderResponse] size: 3
NOK
[HTTP-Client][writeToStreamDataBlock] connection closed or file end (written: 3).
[HTTP-Client][end] still data in buffer (117), clean up.
[HTTP-Client][end] tcp stop
[HTTP] POST... code: 200
[HTTP-Client][returnError] error(-4): not connected
SUCCESS! DATA RECEIVED: 

[HTTP-Client][end] tcp is closed
```

**(B) ESP8266 (running ESPAsyncWebserver)**

```
server.on("/auth/MPUdata", HTTP_POST, [](AsyncWebServerRequest * request) {
AsyncWebServerResponse *response = request->beginResponse(100, "text/plain", MPU_data_received ? "OK" : "NOK");
      response->addHeader("Connection", "Keep-Alive");
      request->send(response);
      if(request->contentLength()){
          Serial.printf("_CONTENT_TYPE: %s\n", request->contentType().c_str());
          Serial.printf("_CONTENT_LENGTH: %u\n", request->contentLength());
    }}, [](AsyncWebServerRequest * request, String filename, size_t index, uint8_t *data, size_t len, bool final) {
            String nameFile;
            nameFile += "/auth";
            if (!filename.startsWith("/")) filename = "/" + filename;
            nameFile += filename.c_str();
            static File fsUploadFile = SPIFFS.open(nameFile, "w+");
            static size_t fileSize = 0;
            
            if(!index){
              Serial.printf("UploadStart: %s\n", filename.c_str());
            }
            if (fsUploadFile) {
                Serial.printf("Continue upload part. Size = %u\r\n", len);
                if (fsUploadFile.write(data, len) != len) {
                    Serial.println("Write error during upload");
                } else {
                    fileSize += len;
                }
            }        
            if(final){
               if (fsUploadFile) {
                    fsUploadFile.close();
                }
                fileSize = 0;
              Serial.println();
              Serial.printf("UploadEnd: %s, %u B\n", filename.c_str(), index+len);
              MPU_data_received = true;
            }      
});
```

(B) HTTP "Serial response" (all steps to write to fsUploadFile is ignored...)
```
_CONTENT_TYPE: multipart/form-data
_CONTENT_LENGTH: 1825
```

hagelric, 2018-03-18:
I did another test creating entire "header" (using boundary identifier) and POST() function instead sendRequest().

`http.addHeader("Content-Type", "multipart/form-data; boundary=----5923c593ca3ddf31ad7b6");`
`int httpCode = http.POST(myContent);`

**myContent is a string:**
_------5923c593ca3ddf31ad7b6_
_Content-Disposition: form-data; name="testfile"; filename="1521369140.htm"_
_Content-Type: text/html_

_[INFORMATION_TEST]_

_------5923c593ca3ddf31ad7b6--_

and this works!

**But...**
I have 2920bytes limit size (if "myContent" has more, data is not received on server side)

How can I use sendRequest() ? ESP8266HTTPClient.cpp doesn't have any "header" manipulation to send boundary (only Content-Length with streamFile.size)

ESPAsyncWebServer doesn't receive the HTTPClient STREAM* and I think that _header_ is the problem...

hagelric, 2018-03-20:
I checked WebRequest.cpp (ESPAsync) and __itemIsFile_ is always "false". How adjust headers on client side (upload side).

PotatoSauceVFX, 2019-05-22:
@hagelric Did you ever figure this out? I am trying to do something similar and with a ESP32 and can't figure it out.

hagelric, 2019-09-23:
> @hagelric Did you ever figure this out? I am trying to do something similar and with a ESP32 and can't figure it out.

I didn't find any solution...

hagelric, 2019-11-26:
Anybody has a solution? I didn't find any method to send large files (or data)...

anctsys, 2019-12-21:
Hi
I search a solution to send binary file to a web server. I not found any solution on the web...this is very funny because only 2 persons are interested by this....

Have you try to use bsae64 , also perhaps we could use wificlient to format the request 

String response = "HTTP/1.1 200 OK\r\n";
response += "Content-Type: image/jpeg\r\n";

anctsys, 2019-12-27:
@hagelric you can send large file via FTP , you can see this contribution
https://github.com/anctsys/ESP32_FTPClient

closed 2020-03-11 by stale[bot]