me-no-dev/ESPAsyncWebServer #340: DefaultHeaders for all http requests?

dphans, 2018-03-22:
How can I send default headers for all http requests?

I'm using ESPAsyncWebServer to make an API routing, testing on local browser call to device "API".
But it required need to set header from response like "Access-Control-Allow-Origin: *".
Current, each response, I create a response from request (inside `<ESPAsyncWebServer instance>.on()`). But I think, do this way is very redundancy. So any idea for me to set default headers for all requests?

I found inside library has a class named `DefaultHeaders`, is it available? I not found this in documents.

Sorry, this is not issue of library, just my problem, you are the best!

penfold42, 2018-03-22:
Seriously ?

https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/README.md#adding-default-headers

Pablo2048, 2018-03-22:
Read the README.md especially section https://github.com/me-no-dev/ESPAsyncWebServer#adding-default-headers ... I'm using this for CORS:
  DefaultHeaders::Instance().addHeader("Access-Control-Allow-Origin", "*");
  DefaultHeaders::Instance().addHeader("Access-Control-Allow-Headers", "content-type");
BTW don't forget to add proper reaction for OPTIONS to your notfound handler.

dphans, 2018-03-22:
OMG this is my fault, sorry. I close this issue now.
Many thanks @Pablo2048 @penfold42

closed 2018-03-22 by dphans

penfold42, 2018-03-23:
*grin* in your defence, the doco is a bit long and easy to miss