me-no-dev/ESPAsyncWebServer #343: Static IP is not achievable in AP mode

sandeeptara, 2018-03-31:
Hi I am using following code
https://github.com/sandeeptara/ESP32-webserver
if I use STA static IP I can not connect to timer server, but if I use AP mode I do not get static IP , reason?

Pablo2048, 2018-03-31:
?? how exactly is async web server library connected with this issue?
BTW: your given example is unreadable to me - don't be too lazy and do some markup formatting...

Phisik, 2018-04-04:
Have you configure DNS server in setup? This code:
```
WiFi.mode(WIFI_STA);
WiFi.hostname(hostname.c_str());
WiFi.config(stationIP, stationGw,  stationMask, stationGw, IPAddress(8,8,8,8));
WiFi.begin(ssid.c_str(), password.c_str());
WiFi.hostByName(TimeServerName, timeServer);
```
gives 
```
[hostByName] request IP for: pool.ntp.org
[hostByName] Host: pool.ntp.org IP: 91.207.136.50
[INFO] System time was set successfully!
```

closed 2019-10-05 by stale[bot]