me-no-dev/ESPAsyncWebServer #348: Design question

J6B, 2018-04-14:
Hi,

I'm working on a project using some temperature sensors (famous ds18b20).
In the loop, every 60seconds, I'm doing a search of sensors, and temperature measurement in order to push values to another system).
Using ESPAsyncWebServer, I'm publishing an URL that should do the same job (sensor lookup, temperature measurement and finally return values in JSON format).

My problem is that there can be conflict because of both stuff share the same OneWire Bus.
I tried to add a Lock mechanism, but ESPAsyncWebServer have no possible delay to wait for unlock from loop.

Does it is possible to have a delay into ESPAsyncWebServer, or, I need to manage things differently?

Thanks a lot

sebaretamozo, 2018-05-21:
I am doing a similar project. I read a temp sensor every 1 second in the main loop() and save it in a global variable. Then, read the variable from ESPAsyncWebServer.

J6B, 2018-05-21:
Thanks everybodies.
I did what @Misiu described. As far as I know, this is the best solution.
I'm reading regularly temperature.
1 Wire and DS18B20 allow to start convert of all temperature simultaneously, I'm saving time.
and ESPAsyncWebServer return those when requested

closed 2018-05-21 by J6B