me-no-dev/ESPAsyncWebServer #351: Best way to use bootstrap, bootstrapjs and jquery?

zurbran, 2018-04-19:
Hello i want to add the bootstrap.min.css, bootstrap.min.js and jquery.js to my index.html, i cant find a good way to do it. I put them on the flash mem like this:
String bootstrapCSS(F("All the boostrap.min.css string here with all the scapes and stuffs in order to be valid"))'
 Also y made variables for de Jquery and boostrapJS but when i call the method:
server.on("/home", HTTP_GET,[](AsyncWebServerRequest *request) {
		request->send(200, "text/html", ???????(i dont know how to put all together here) );
	});
If someone did it, it will be really helpful.
Thanks in advance.
PD: I tried to make a huge string combining all of them but when i load it with the browser the full html with css and js dont want to load completly, instead i get only a chunk of the whole string displayed in the browser without any format at all.

penfold42, 2018-04-19:
Check out https://github.com/forkineye/ESPixelStick

It does what you’re looking for

JamesNewton, 2018-04-30:
You can also do multiline files like this:

````
const char PAGE_Root[] PROGMEM = R"=====(
your huge file
with 
many lines

)=====";

void handle_root(AsyncWebServerRequest *request) {
  request->send(200, "text/html", PAGE_Root);
}
````

and then in the setup:
````
  server.on ( "/", HTTP_GET, handle_root  ); //main web page in Page_Root.h
````

closed 2019-10-05 by stale[bot]