me-no-dev/ESPAsyncWebServer #352: What happens if I block in a handler?

dpetican, 2018-04-19:
I want to block in a handler to wait for a response from Serial to put into the body of the response.  Its okay that that I won't be able to accept any connections.  Is espasyncwebserver cooperative or pre-emptive? Do handlers run to completion or yield? Meaning if I do block in the handler will the loop stop running?  Which is okay for me. I will have a timeout in case I receive no response from the Serial port and return a 404 error.  Thanks.

dpetican, 2018-05-03:
Actually I've been able to block with a while loop for up to 5+ seconds if I kick the watchdog.  There are no detrimental effects.  The webserver accepts requests after the handler returns.  Delay does nothing probably because it yields.  However, what I can't seem to do is actually read from serial.  This causes an immediate watchdog reset, probably again because of yield.

dpetican, 2018-05-03:
This works but I can't read a variable number of bytes:

` AsyncWebServerResponse *response = request->beginResponse(Serial, "text/plain", 50);
// Here is where I will send a command over serial that I want to get a response to 
unsigned long startTime = millis();
 while (millis() - startTime < SERIAL_TIMEOUT) {
   wdt_reset(); // kick the doggie!!
   Serial.println(millis() - startTime);
 }
  response->addHeader("Server","ESP Async Web Server");
  request->send(response);
}
`

joaocarlosss84, 2019-05-17:
Hi, is it possible that you store that variable from Serial and display it later at the webpage?
Or maybe you can use websocket, so the user will see the webpage and when you have the result you can use the websocket to update the HTML on the fly.

closed 2019-10-05 by stale[bot]