me-no-dev/ESPAsyncWebServer #353: WiFi.SoftAP() : Connection refused

psykokwak-com, 2018-04-30:
Hello,
Following this code, if the WiFi STA connection failed, it fallbacks to AP mode.
But the server returns "connection refused". (TCP connection refused).

` void networkWiFiInitialize()
{
  WiFi.mode(WIFI_STA);
  WiFi.begin(_networkESSID, _networkESSIDPwd);
  WiFi.setAutoConnect(true);
  WiFi.setAutoReconnect(true);
  while (WiFi.waitForConnectResult() != WL_CONNECTED)
  {
    // Act as an AP if no existing AP connexion is made
    IPAddress apIP(192, 168, 1, 1);
    IPAddress apNetMsk(255, 255, 255, 0);

    WiFi.mode(WIFI_AP);
    WiFi.softAPConfig(apIP, apIP, apNetMsk);
    WiFi.softAP("MyAP");
  }`

If I remove the WiFi station part of the code to directly starting as AP mode, it works : 
` void networkWiFiInitialize()
{
  //WiFi.mode(WIFI_STA);
  //WiFi.begin(_networkESSID, _networkESSIDPwd);
  //WiFi.setAutoConnect(true);
  //WiFi.setAutoReconnect(true);
  //while (WiFi.waitForConnectResult() != WL_CONNECTED)
  {
    // Act as an AP if no existing AP connexion is made
    IPAddress apIP(192, 168, 1, 1);
    IPAddress apNetMsk(255, 255, 255, 0);

  
    //WiFi.mode(WIFI_AP);
    WiFi.softAPConfig(apIP, apIP, apNetMsk);
    WiFi.softAP("Sentinel OTA Upgrade");
  }
}`

tarzan115, 2018-05-02:
this is not right place for your question. but I think you need this https://github.com/tzapu/WiFiManager

debsahu, 2018-05-04:
@tarzan115 WiFiManager is not compatible with ESPAsyncWebServer, recommend users to look at [AsyncWiFiManager](https://github.com/alanswx/ESPAsyncWiFiManager).

psykokwak-com, 2018-05-10:
Ok. I have to use WIFI_AP_STA instead of WIFI_AP in WiFi.mode.

closed 2018-05-10 by psykokwak-com