me-no-dev/ESPAsyncWebServer #359: latest Arduino ESP32 core + latest mkspiffs 0.2.3 & latest AsyncWebserver no static files are served

sdsbjn, 2018-05-09:
HI,
I don't get SPIFFS to run. I can upload without errors, but with AsyncWebserver no static SPIFFS file are served. Any ideas ? The same code (without partition) worked well with Wemos D1,R1 mini

**Name, Type, SubType, Offset, Size, Flags**
nvs, data, nvs, 0x009000,0x005000,
factory, app, factory, 0x010000,0x0C0000,
eeprom, data, 0x99, 0x0D0000,0x001000,
spiffs, data, spiffs, 0x0D1000,0x320000,

jesse-schein, 2018-05-09:
No problems here on esp32.

Mr-Blonde, 2018-06-03:
How did you upload files into SPIFFS? For me, the ESP8266 Sketch Data Uploader Plugin is currently not working. There is no error message, but no files are copied into the flash (Wemos D1 mini Pro). 

To verify, what's actually in the flash you can use:

> String str = "";
> Dir dir = SPIFFS.openDir("/");
> while (dir.next()) {
>     str += dir.fileName();
>     str += " / ";
>     str += dir.fileSize();
>     str += "\r\n";
> }
> Serial.print(str);

([source](https://www.esp8266.com/viewtopic.php?t=8459))

closed 2019-10-05 by stale[bot]