me-no-dev/ESPAsyncWebServer #362: No webpage with SPIFFS

axoulc, 2018-05-16:
Hello,
I upload my html file in SPIFFS mem in "www" folder and I can see this file with the help of SPIFFS functions.
This html file in also in a PROGMEM variable only for test.
`request->send(200, "text/html", HTML);` This line work great !
`request->send(SPIFFS, "/www/index.htm", "text/html");` But nothing with this line even if it is the same HTML code.
ESP32 return no log about.
Whole code : [https://github.com/axoulc/Logger/blob/master/main/wifi.cpp](url)

If someone can help me.
Thanks in advance
Axel

drschlaumeier, 2018-05-18:
SPIFFS.begin() ???

axoulc, 2018-05-18:
Hi, check wifi.cpp and I init spiffs before read in.

drschlaumeier, 2018-05-18:
ahh, sorry...
I was using request->send(SPIFFS, "/jquery.min.js.gz"); to read from SPIFFS but ESP32 is crashing #361 .

Did you check if file exists?
if (SPIFFS.exists("/jquery.min.js"))  ...

closed 2019-10-05 by stale[bot]