me-no-dev/ESPAsyncWebServer #372: ESP32 & ESPAsyncWebServer servestatic & vfs_fat: open: no free file descriptors

r-anwar, 2018-06-28:
Hi,

I'm delivering an angular5 application through your (great!) webserver, nevertheless when serving multiple files simultaneously, e. g. opening the website from different browsers at the same thime, I get following error. "vfs_fat: open: no free file descriptors".

May somebody can help?

Following you will find the complete Backtrace:

E (36605) vfs_fat: open: no free file descriptors
[E][vfs_api.cpp:238] VFSFileImpl(): fopen(/sd/www/main.f4aded1c101d0e50ff70.bundle.js) failed
CORRUPT HEAP: Bad tail at 0x3fff238b. Expected 0xbaad5678 got 0xbaad5600
assertion "head != NULL" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/./multi_heap_poisoning.c", line 201, function: multi_heap_free
abort() was called at PC 0x400f1787 on core 1

Backtrace: 0x4008d94c:0x3ffeccd0 0x4008db27:0x3ffeccf0 0x400f1787:0x3ffecd10 0x4008d65d:0x3ffecd40 0x4008412e:0x3ffecd60 0x40085e0d:0x3ffecd80 0x4000bec7:0x3ffecda0 0x400defe7:0x3ffecdc0 0x400ddab6:0x3ffece10 0x400dc7e6:0x3ffece50 0x400dada3:0x3ffece90 0x400dd263:0x3ffecf80 0x400dd309:0x3ffecfc0 0x400dd561:0x3ffed010 0x40139681:0x3ffed030 0x401398ad:0x3ffed080

r-anwar, 2018-06-28:
Addition Information: Sometimes, same scenario, its crashes with following exception:

Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited)
. Exception was unhandled.
Core 1 register dump:
PC      : 0x3f401b84  PS      : 0x00060d30  A0      : 0x80138ca5  A1      : 0x3ffed300  
A2      : 0x3ffee860  A3      : 0x00000000  A4      : 0x3ffee7e4  A5      : 0x00000000  
A6      : 0x3ffed31c  A7      : 0x3ffeb400  A8      : 0x8013897d  A9      : 0x3ffed2f0  
A10     : 0x3ffee860  A11     : 0x3ffed30c  A12     : 0x3ffed308  A13     : 0x00000000  
A14     : 0x3fff32f8  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x00000014  
EXCVADDR: 0x3f401b84  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x3f401b84:0x3ffed300 0x40138ca2:0x3ffed330 0x40138cff:0x3ffed360

r-anwar, 2018-06-28:
Or this exception:

assertion "heap != NULL && "free() target pointer is outside heap areas"" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/./heap_caps.c", line 261, function: heap_caps_free
abort() was called at PC 0x400f179b on core 1

Backtrace: 0x4008d94c:0x3ffecef0 0x4008db27:0x3ffecf10 0x400f179b:0x3ffecf30 0x40084127:0x3ffecf60 0x40085e0d:0x3ffecf80 0x4000bec7:0x3ffecfa0 0x40110e99:0x3ffecfc0 0x40110f5b:0x3ffecfe0 0x401389b6:0x3ffed000 0x40138c8c:0x3ffed040 0x40138cff:0x3ffed070

r-anwar, 2018-06-28:
And this exception:

CORRUPT HEAP: Bad head at 0x3ffec3d8. Expected 0xabba1234 got 0x3ffec698
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Core 1 register dump:
PC      : 0x4008cca0  PS      : 0x00060334  A0      : 0x8008c076  A1      : 0x3ffdc970  
A2      : 0x3ffe3284  A3      : 0x3ffdd068  A4      : 0x3ffc11c8  A5      : 0x00000001  
A6      : 0x00060323  A7      : 0x00000003  A8      : 0x3ffdd068  A9      : 0x3ffdd068  
A10     : 0x00000007  A11     : 0x00000007  A12     : 0x3ffec3e0  A13     : 0x00000001  
A14     : 0x00060323  A15     : 0x3f406264  SAR     : 0x00000004  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffe9  

Backtrace: 0x4008cca0:0x3ffdc970 0x4008c073:0x3ffdc990 0x4008aeb3:0x3ffdc9b0 0x4008af1d:0x3ffdc9f0 0x40081375:0x3ffdca10 0x4008145d:0x3ffdca40 0x4014ddbc:0x3ffdca60 0x400f06cd:0x3ffdca80 0x4000bd83:0x3ffdcaa0 0x4000117d:0x3ffdcac0 0x400592fe:0x3ffdcae0 0x4005937a:0x3ffdcb00 0x40058b69:0x3ffdcb20 0x400fb1e9:0x3ffdcb50 0x400fd285:0x3ffdcb80 0x400f70c5:0x3ffdce40 0x400f1670:0x3ffdce90 0x4008d65d:0x3ffdcec0 0x4008412e:0x3ffdcee0 0x40085e0d:0x3ffdcf00 0x4000bec7:0x3ffdcf20 0x4010ac59:0x3ffdcf40 0x4010ad1b:0x3ffdcf60 0x40133885:0x3ffdcf80 0x40107049:0x3ffdcfa0

Core 0 register dump:
PC      : 0x4008b752  PS      : 0x00060034  A0      : 0x8008c125  A1      : 0x3ffc0510  
A2      : 0x3ffc11a8  A3      : 0x3ffc0588  A4      : 0x00000008  A5      : 0x3ffdd9c4  
A6      : 0x00000018  A7      : 0x0b46ed05  A8      : 0x0000cdcd  A9      : 0x0000cdcd  
A10     : 0xb33fffff  A11     : 0x0000abab  A12     : 0x00000000  A13     : 0x00000001  
A14     : 0x00060023  A15     : 0x00000000  SAR     : 0x00000019  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x4008b752:0x3ffc0510 0x4008c122:0x3ffc0530 0x4008aca1:0x3ffc0550 0x40086872:0x3ffc0570 0x40085fc9:0x3ffc05c0 0x40089170:0x3ffc05e0 0x40081891:0x3ffc0610 0x400f2feb:0x00000000

r-anwar, 2018-07-04:
Is it allowed to use the serveStatic method with SD? 

I'm serving my files like this:

```
server.serveStatic("/frontend/", SD, "/www/").setCacheControl("max-age=31536000");
server.rewrite("/", "/frontend/index.html");
```

closed 2019-10-05 by stale[bot]

teq-nn, 2023-06-16:
Hey @r-anwar, did you manage to resolve this issue somehow? I am having the same issue during first time page load.

Web-cloud-warrior, 2023-08-15:
For anyone else with this issue, you can change the begin command to increase the number of files the ESP32 can open. Most web clients will try to open 8 or so. The SD.Begin defaults to 5 files max. Hence the errors. the default for the SD is 5 files.

Web-cloud-warrior, 2023-08-15:
try

  if (!SD.begin(chipSelect,SPI,4000000U,"/sd",**10**U,false)) {  
    Serial.println(F("Card failed or not present, no SD Card data logging possible..."));
    SD_present = false; 
  }

teq-nn, 2023-09-22:
Thanks @Web-cloud-warrior, that did the trick. 
In the meantime I managed to reorder the frontend html, so that the smaller files are loaded before the larger ones. This ensures that there are no more than 5 opened files at once. Though this might be heavily depending on the application.